OpenHD/OpenHD #159: Move some boot up functionality to a USB stick

Beegad42, 2019-09-16:
It is a real hassle to have to pull an SD card to change a setting or battle with getting an android app to connect to a WiFi hotspot that is disabled or try and get the ground to work on an Ethernet connection that used to work but now it doesn't.

Ii is proposed that the .profile script first check if there is a main.sh on a USB stick and run that instead of the one on the image. An experienced user could tweak that script e.g. to set up a test image or it could be generic for dummies like me. The dummy version could present a menu e.g.

Press 1 to copy settings file from this USB stick to image.
Press 2 to invoke RemoteSettings to transfer settings from ground to air.
Press 3 to enable Ethernet hotspot.
Press 4 to enable Wifi Hotspot.
Press 5 to start DHCP client on Ethernet and show IP address obtained.
Press 6 to copy OSD config from USB to image and test
Press 7 to backup important settings to USB.
Press 8 to copy debug log from image to USB.
And so on...

All of that stuff/scripts/complexity/unreliability should then be deleted from the Open.HD image and moved to a separate GIT project. 
If a user wants to update their settings they can do this with a text editor on the USB stick, plug this in and follow the prompt.

RespawnDespair, 2019-09-26:
It is an interesting proposal. Pi zero doesn't leave a usb port for this, but the ground pi could. It raises the point of re-thinking the way we handle the startup now. Perhaps others can chime in and pitch ideas on this subject?

Beegad42, 2019-09-27:
I was thinking of this for GroundPi, if ground is in a known state i.e. with Wifi AP enabled and air is updated to detect this (wifi client) then the state/settings of ground and air can be controlled by booting ground from USB script.

RespawnDespair, 2020-12-04:
Ease of development hasn't been a problem for a while now. And changing settings has never been easier since QOpen.HD.
Some of these suggestions might come around again, but for now I will close the issue.

closed 2020-12-04 by RespawnDespair