OpenHD/OpenHD #160: Ditch tx_measure in favour of tx_rawsock (simplication of code base)

Beegad42, 2019-09-16:
home/pi/wifibroadcast-base/tx_measure is used to calculate a maximum video rate that the underlying Wifi hardware can support.  It basically fills the outgoing raw socket with packets and times the rate at which they were written to that socket.

The code itself looks like an early version/copy of tx_rawsock hacked to do the job. This creates two code bases that must be kept in sync. That is any new features added to tx_rawsock.c (e.g. MCS) must also be added to tx_measure. 

It is proposed that tx_rawsock be used instead of tx_measure and tx_measure be discarded.  The work involved is to add a command line parameter to tx_rawsock flagging that it should measure throughput and copy some of the fifo and measurement logic across from tx_measure.

htcohio, 2019-09-17:
Makes sense. 

Are you by chance capable of putting together some proposed changes to test out?

I also I think that we need some structure to the measurements for the alternative IP USB camera that is still under development.

But, part of that currently utilizes the SVPCOM version of UDP h.264 so I'm not sure if that is structured differently.

From what I can tell, TX_rawsock itself is more reliable and results in less jitter with the video playback.

Beegad42, 2019-09-18:
I am happy to code this up once you have development merged back in to main

zipray, 2019-09-18:
The measurement function only runs once at startup, providing the camera coding speed. Does the combination mean that the fifo and measurement logic across tx_measure will always be running?

RespawnDespair, 2019-09-26:
> Does the combination mean that the fifo and measurement logic across tx_measure will always be running?

No i think the proposal is to have 1 program that can be run once at startup to perform the measurements and as the regular tx_rawsock that remains running. At least that is what i understand from the discussion.

PR #249 (2020-01-30): Remove old tx_measure code
Replaced by additions to tx_rawsock in #246

Closes #160
commit ff199ae5 (2020-01-30): Remove old tx_measure code

Replaced by additions to tx_rawsock in #246

Closes #160
closed 2020-01-30 by steveatinfincia