OpenHD/OpenHD #161: Lat-Long first two digits are not being displayed in latest OpenHD image of 9th Sept

VU2ATX, 2019-09-17:
Lat-Long first two digits are not being displayed in latest OpenHD image of 9th Sept, pls see the image attached
[image]

htcohio, 2019-09-17:
Hi,

Yeah, we included that in the release notes, you can disable it in the open HD settings. Txt file.

We implemented that to protect users posting video identifying their exact location. 

The first 2 Lat/Lon numbers can be obtained easily googling your location to "fill in the blanks". 

*The first 2 numbers only change over long distances..

E.g. I didn't do the math but, It is 100 miles + for Lat/Lon

VU2ATX, 2019-09-17:
Thanks, I did not notice that. I am just checking it and update

VU2ATX, 2019-09-17:
Yes, its working, thanks

closed 2019-09-17 by VU2ATX