OpenHD/OpenHD #164: Skip FEC feature of  tx_rawsock should be configurable (i.e. in settings.txt)

Beegad42, 2019-09-20:
The logic of tx_rawsock uses a microsecond timer to measure how long it took to write the most recent FEC block of packets to the raw socket. Under default settings an FEC block  would be 8 x 1024 data packets plus 4 FEC packets, ballpark 12k.

If the time taken exceeds a ballpark threshold the logic will discards exactly 4 FEC packets from the next FEC block transmitted. The number of FEC packets discarded is hard coded and does not take in to account how many FEC packets are actually in a block.

Nor does the logic take into account that a task switch during transmission would greatly distort the time measurement, there is no averaging/sanitizing or discards of extraordinary measurements.

Put simply the feature is probably useless on a single core CPU (e.g. Pi zero). The logic looks experimental, needs improving therefore it  should be a configuration parameters such that uses can trial the feature rather than having no choice.

pilotnbr1, 2019-10-03:
Now that dev merged into master and things have settled down, any chance you will work on this?

RespawnDespair, 2020-12-04:
After checking with the other developers this will be irrelevant/implemented in the 2.1 branch and is not something we want to change in the current 2.0 since it might impact stability.

This issue will therefore be closed, but thank you again for your insights!

closed 2020-12-04 by RespawnDespair