OpenHD/OpenHD #168: RTL8812AU TX and RTL8814AU RX combo seems flawed with OpenHD2.0 RC1.1

KRUAndu, 2019-09-27:
I have:
- **TX** RPi Zero (ASUS USB-AC56 w/ RTL8812AU), OpenHD2.0 RC1.1
- **RX** RPi 3B+ (ALFA AWUS1900 w/ RTL8814AU), OpenHD2.0 RC1.1
- only telemetry and video is used, no RC.

The issue is when TX has the 8812AU card, there is no camera image received on the GroundPI, although telemetry info is received + CPU usage and temps...

Now when I swap the cards (RX is 8812AU and TX 8814AU), I get video and telemetry. But I don't want to swap the cards on my quad because of this. Supposedly this new version fixed some RSSI issues with the RTL AC cards? And introduced some new bugs too... :)

Just for reference, my current configuration (as above) worked with this image "PRESET FOR RTL8812AU_STBC-MCS2-Open.HD-2019-08-10.iso" with no issues.

Can this issue be linked to this: https://github.com/HD-Fpv/Open.HD/issues/139 ?

I'm willing to help.

KRUAndu, 2019-09-27:
This is the config file if you need it: [https://pastebin.com/vcxmFVru](url)

KRUAndu, 2019-09-27:
@htcohio, your proposed fix on issue #139 worked for me, reversing the changes on tx_functions.sh from commit https://github.com/HD-Fpv/Open.HD/commit/1696cc848e974aa4ed4201c7baa08bd2377efe82 got the image back on my RX.

htcohio, 2019-10-08:
Hey, that is interesting to know, I can say that the Realtek cards and drivers are a little bit of a mystery and have a lot of Frankenstein components as mentioned by the aircrack team...

I was actually under the impression that the 8814au was never really working recently so, it's good to know that rolling back Ziprays patch (which was intended to pair up the correct frame type i.e. RTS or Data for the video and RC use cases... 

I will close this issue as it's working for you but please keep us updated on the stability of the setup;):

closed 2019-11-03 by pilotnbr1