OpenHD/OpenHD #169: ENHANCE PICAPTURE COMPATIBILITY

roque-canales, 2019-09-27:
picapture board no send data as picamera when in .profile and global_functions.sh send command:
CAM=`/usr/bin/vcgencmd get_camera | nice grep -c detected=1`

to determine if camera is connected.

I tryed to replace in the two script files :
CAM=`/usr/bin/vcgencmd get_camera | nice grep -c detected=1`
per
CAM="1"

without succes….. OHD continuous to work as if I Don't tuched anything….
I recheck the fil to make sure that modification was drone, and it is.

So there is compilation needed, or copy of thèses files in another hiden place ?

Does anybody have and idea to desactivate this detection for picapture users ?

thank you so much for help us.

Regards.

Beegad42, 2019-09-27:
Did you delete the air.txt file that you created as a first attempt to get around the issue?

roque-canales, 2019-09-27:
Yes absolutely!

closed 2023-01-31 by raphaelscholle