OpenHD/OpenHD #172: OSD Development

pilotnbr1, 2019-10-03:
Dumping all the enhanced OSD bugs and wishlist stuff here-

BUGS
- Home RADAR  correction in the southern hemisphere should be reversed
- LAT LON reverse icons

ENHANCEMENT
- clock icon for timer
- rolling packet loss/declutter
- spacing for speed and alt trends
- box on speed and alt
- break out mah
- bat cells display individual volts
- hud and pipper
- complex compass
- LTM throttle
- sea level icon with water
- cts declutter

WISHLIST (probably wont happen)
- compass on AHI
- home icon on compass
-lat lon declutter reappear for rtl
- angle indicators declutter
- HDOP is its own element

pilotnbr1, 2019-10-08:
Very interested in smoothing the osd. Some elements are noisy and choppy. Particularily speed and altitude. My proposal is to slightly delay some of those numbers and then interpolate to get a value across a moving average.

questions are what is the current framerate of the render and how much cpu will be used if that draw cycle is increased

ABLomas, 2019-10-08:
One more "wishlist item" - fixed AHI.
Current AHI is not so usefull - it's small (can be scaled bu then other lines scale too) and main horizon line gets out of screen quickly. When flying blind (night, clouds, fog) it's very hard to follow pitch angle by looking at numbers (which also jump on screen at the same time) and roll on small lines.
Suggestion: make AHI one thick line. No "secondary lines" required, just one bold line, also wide.
Roll should follow horizon (keep as-is), pitch should be scaled by ~4x times (so 90° angle will clip out of screen but < 90° it will be still on screen, partially if roll angle !=0)
Pitch angle can be displayed by single number on line, this way it would be much easier to climb at constant angle.
On-screen object movement will be decreased a lot (by reducing disappearing lines amount) reducing distraction. One wide line allways visible on screen is much easier to follow when flying only using OSD.

P.S. used all of them on analog a lot - one wide line on matrixpilot, many lines (like current open.hd) on rvosd, one simple line is much better IMO.

pilotnbr1, 2019-10-10:
ABlomas I AGREE. Will add zipray ahi and make it default. Will add bore fixed boresight as well.

Inav not playing well with mavlink sys status messages... report that these alarms don’t ever shutoff
some alerts stay on INAV all time (with GPS lock, etc): 11, 12, 15, 16

pilotnbr1, 2019-10-15:
Artificial Horizon has several changes now.

- single thick bar by default
- does not disappear with extreme pitch but is also less sensitive to pitch
- clipping was used to define box where ahi is visible so it programatically does not limit its draw area. Clipping is less desireable as it might unload gpu and transfer load to cpu according to openvg forum
- At extreme pitch roll seems overly sensitive
- Flight path vector is now drawn with ahi. It is not a true vector. Due to AHI being non-conformal (does not attempt to represent the AHI overlayed on video where the actual horizon is) a true vector does not have benefit. FPV just displays m/s data in vy and vz axis

Beegad42, 2019-10-15:
link to latest OSD dev image
https://drive.google.com/file/d/1m9KsbdIYnTU_I5peiS-x_ULgXSuWOH3J/view?usp=sharing

ABLomas, 2019-10-17:
In latest image, linked in previous comment - If:
`#define REVERSE_ALTITUDES`
is set to "true" (default: false), OSD will fail to boot.
This is probably important, as i do not know another way to have simple numeric-only AGL altitude display without ladder.
If it is set to "false", numeric display will display only MSL (then in flight we need to calculate alt above ground).

ABLomas, 2019-10-23:
> Inav reported not detecting arm

I was wrong, DECLUTTER works fine with INAV.

AHI  works very well, example in flight: https://youtu.be/IjbGU5B2xF0
What that yellow triangle is doing - nobody knows =)

pilotnbr1, 2019-10-23:
So yellow triangle is what the creator labeled “home radar”. Observing it’s behavior on recent videos it appears that you just “steer” the triangle to the center of the bore sight to get the vehicle “home”.

careyer, 2019-10-24:
Yes, the home radar thingy is really useful if you lost orientation and or video and want to return home without automatic RTH.  Pretty much steer the yelloe triange to the center of the screen like in a computer game and you find your way back home =)

maguet11020, 2019-11-04:
Any possibility to get it work with pixhawk 4 ? ( no flight mode and rssi ) 
why is it limited to 2 antennas ?

pilotnbr1, 2019-11-10:
> Any possibility to get it work with pixhawk 4 ? ( no flight mode and rssi )
> why is it limited to 2 antennas ?

Pixhawk 4 has revealed differences between ardupilot and px4 with base modes and custom modes in the heartbeat message. Right now custom modes are parsed... Solution is to change everything and read something other than custom modes from heartbeat OR learn the px4 custom mode numbers and add them to mavlink.c or render.c (hacky easy fix)

Not sure what you mean about 2 antennas?.. maybe ask that in telegram or rcgroups

I will probably fix the reverse-altitudes bug and the yaw pointer- I created those problems (try to clean up after myself). As for more development on this osd I am shifting my efforts to qopenhd as it sounds like it will be the new default standard

maguet11020, 2019-11-10:
Hi,

To get it work you have to use arduplane on px4

steveatinfincia, 2020-12-07:
Most of the things in this issue should be resolved now or planned separately in QOpenHD issue tracker.

closed 2020-12-07 by steveatinfincia