OpenHD/OpenHD #173: IP Camera based on Hi351x family not working with the latest image of OpenHD

VU2ATX, 2019-10-03:
**IP Camera based on Hi351x family not working with the latest image of OpenHD (Open.HD_V2,0_RC1,1-2019-09-13)**

Hi, configured the IP camera as per required, no success so far

User Name of IP Camera Login: admin
Password: nil (no password, kept it blank)

Created "air.txt" in AirPi

I have Pi3B+ as GroundPi and Pi3B as AirPi

Connected the IP Camera based on Hi3518 to the Pi via LAN cable.

IsBandSwicherEnabled=1
SecondaryCamera=IP

IPCameraHiRes="gst-launch-1.0 rtspsrc location="rtsp://192.168.0.10:554/user=admin&password=&channel=1&stream=0.sdp" latency=0 ! udpsink port=5600 host=127.0.0.1"

IPCameraLowRes="gst-launch-1.0 rtspsrc location="rtsp://192.168.0.10:554/user=admin&password=&channel=1&stream=1.sdp" latency=0 ! udpsink port=5600 host=127.0.0.1"
[image]

best4username, 2019-10-04:
same thing with me 
after digging :
found that when svpcom wfb_tx start the rssitx exit itself
the gstreamer send video to 5600 and wfb_tx only show:
listen on port 5600
but wfb_rx only gives continue lines with this  :
 1459454  PKT  0:0:0:0:0:0
 1459455 PKT  0:0:0:0:0:0
 1459456  PKT  0:0:0:0:0:0
tried to down wifi inteface and config it again manuly with no sucsses
there no error on dmesg 
but every thing else work out the box thanks a lot

user1321, 2019-10-04:
Hi.
Are you sure that RTP url is correct and it is same as for Hi3516? 
Are you able to view video using... for example VLC player in case of Hi3518 and url:
rtsp://192.168.0.10:554/user=admin&password=&channel=1&stream=1.sdp
Or it was working for you in the past with old OpenHD image`s and exactly that camera?

user1321, 2019-10-04:
Also, from your first post I see missing symbol in config, don`t know, maybe it is just typo or something...

VU2ATX, 2019-10-05:
> rtsp://192.168.0.10:554/user=admin&password=&channel=1&stream=1.sdp

Hi, 
Yes, RTP url is correct and it is same as for Hi3516.
I am able to view the video via CMS on PC.
I have tested it for the first time only on latest image (Open.HD_V2,0_RC1,1-2019-09-13) of OpenHD.

best4username, 2019-10-05:
the URL is correct for me also, Tried to forawrd stream from 5600 to my pc and view it with gst it worked fine
I think problem with svpcom wfb_tx or rx ?

VU2ATX, 2019-10-05:
Has anyone tested and seen the IP camera working ok with OpenHD any image order or newer ?
If yes, then which image ?

user1321, 2019-10-05:
> same thing with me
> after digging :
> found that when svpcom wfb_tx start the rssitx exit itself
> the gstreamer send video to 5600 and wfb_tx only show:
> listen on port 5600
> but wfb_rx only gives [...]

To exclude any problems with TX\RX keys, wrong freq, etc... Can you please try to use USB camera mode?
There is a videotestsrc pipeline:
#gstreamer videotestsrc:
USBCamera="gst-launch-1.0 videotestsrc ! 
So, if you did everething correct, please, try to change 
___________________
# Values: No, IP, USB
SecondaryCamera=No
_________________________
from IP to USB:
SecondaryCamera=USB
And look if you will able to see test video image on ground side.

user1321, 2019-10-05:
> Has anyone tested and seen the IP camera working ok with OpenHD any image order or newer ?
> If yes, then which image ?

Yes, it is works for me. But, with Hi3516.
I may share old image that I am using only for myself. It is a bit old.
Actually, I did`t saw any critical changes inside OpenHD repository for last month or two that could break camera.
Actually, can you please confirm that IP camera working with VLC, not CMS (I suspect you don`t type any pipeline to CMS, unlike VLC)

user1321, 2019-10-05:
>>Created "air.txt" in AirPi
Also, one more crazy idea. I actually don`t know why it not working for you, so it is time for crazy idea`s) :

Are you sure that on air side "air.txt" is actually "air.txt" and not "air.txt.txt" ? That can happen in case if windows (if you use windows) is not displaying file extensions.
In any case you can verify that with "SecondaryCamera=USB" test and videotestsrc pipeline.

VU2ATX, 2019-10-06:
Hi, bottom line is, I got success ! IP camera (Hi3518 based) switching via toggle of a switch CH7 on transmitter is also happening, I can see well, both the videos in QGCS custom.
I had to do some change in the IP Camera pipeline;
IPCameraHiRes="gst-launch-1.0 rtspsrc location=\"rtsp://192.168.0.10:554/**11**/user=admin&password=&channel=1&stream=0.sdp\" latency=0 ! udpsink port=5600 host=127.0.0.1"
IPCameraLowRes="gst-launch-1.0 rtspsrc location=\"rtsp://192.168.0.10:554/**11**/user=admin&password=&channel=1&stream=1.sdp\" latency=0 ! udpsink port=5600 host=127.0.0.1"

[image]

best4username, 2019-10-06:
Try USB camera mode with no success .
rssi & cpu temp work only for 5 sec after Air Pi start then it stuck ??
How to disable what VU2ATX say about 

> switching via toggle of a switch CH7 on transmitter is also happening, I can see well, both the videos in QGCS custom.

Can i switch it without remote ?

best4username, 2019-10-06:
Found Solution :
The problem with "-t" argument in "wfb_tx" command line just removed it and try I solved my problem ( Ip camera stream recived on RX with small latency ) then i fix it in " RemoteSettings/Air/TxSecondaryCamera.sh ".
I think it related to my wifi chipset or kernel driver !

But RX rssi and data rate and error counter dont work anymore ??
 Back to raspberry camera every thing work , try Ip camera no RX rssi or data rate ??

thanks a lot for quick response

user1321, 2019-10-06:
Interesting...
As for SVPcom and "-t", that param is not present in original SVPcom tx program, it was added a year ago or so to fix problems with ath9k card`s (TP-Link wn722 - AR9271).
Ath9k driver stuck (hung) in case if in same time transmitted RTS (request to send) and DATA type packets.
So, idea was to let SVPcom transmit both types of packet`s dependig on situation. ( in one moment of a time only RTS or DATA)
That was tested with ath9k and 8812AU type of cards. However, was not tested with ralink ( I don`t have it)
Good find! (if you can notify someone in main support telegram chat about that - that will be good.)
Probably it is a good idea to add that to OpenHD scripts to automatically set appreciated "-t" values depending on Ralink card also in additional to ath9k and 8812AU.
____________________________________________________________________
As for RSSI and IP camera. It is not supported. Someone smart have to add code to implement that.
In my own repository I have ugly and wrong implementation of RSSI for IP camera. It provide incorrect data for lost packets, but, show RSSI. You may look at it and use if you ok with it.
Or, implement own version - better.

best4username, 2019-10-06:
What shock me in VU2ATX video RSSI & data rate appear to be working !!
if he is not switching between to camera ( when raspberry it work and when IP it stuck in last number ) ??

user1321, 2019-10-06:
>>we can make svpcom rx push rssi and data rate values to sheared mem where osd can read it, I'll try
RSSI is ok, there are other data types that are not correspond directly between WFBC RSSI_TX process and SVPcom -  like lost packets, etc... Please, check "wrong workaround. IP camera RSSI".

As for youtube link.. yes, that is strange. I did`t saw RPi camera.. and I not sure what he actually is switching.
It can be 5\10\20 MHz band in case of ath9k.
Or it can be RPi camera and IP camera.
Probably only @VU2ATX can answer.

In any case if you will have new`s about RSSI and IP camera (maybe it is already implemented by someone. It is good to ask in telegram group), please, let me know.
Good luck.

user1321, 2019-10-06:
>>we can make svpcom rx push rssi and data rate values to sheared mem where osd can read it, I'll try

PS. it is already implemented to save to shared memory.
Only question is correspond data types like lost packets, lost block`s.

[image]

Or, maybe there already other implementation for that.

user1321, 2019-10-06:
https://www.youtube.com/watch?v=lp5FwNpCIFY&feature=youtu.be

closed 2019-10-06 by user1321

VU2ATX, 2019-10-06:
@user1321;

Or it can be RPi camera and IP camera.
Probably only @VU2ATX can answer.

In the video, I am switching between PiCam and IPCam with toggle switch on Transmitter which is connected ground Pi via AeroSIM-RC flight simulator dongle.