OpenHD/OpenHD #174: Discussion - add timestamp to packets by adding RTP header to Video and Telemetry packets.

Beegad42, 2019-10-06:
Protocols for streaming video over lossy data links have been standardized via RFC3550/Realtime transport protocol. 

https://en.wikipedia.org/wiki/Real-time_Transport_Protocol
https://tools.ietf.org/html/rfc3550

Frame Error Correction over those links have also been standardized in RFC2733
https://tools.ietf.org/html/rfc2733
As discussed here:https://pdfs.semanticscholar.org/2345/85a301d07dd76af141b1f29377d42da020da.pdf

Most video encoders support RTP encapsulation directly and decoders (such as in QGC) support playing the RTP stream directly.

It is proposed to add a Setting "UseRTP" in which case Airpi will start a different version of tx_rawsock that encapsulates raw video in RTP (or reads RTP from UDP generated by encoder) and starts a different rx on Ground pi to decapsulate for HDMI or pass on via encapsulation in UDP for QGC. 

If the same is applied to telemetry the timestamps in the RTP header can then be used to detect video latency at groundpi.

This feature would add about 12 bytes to each packet (i.e. 1%).

pilotnbr1, 2019-11-03:
This looks really interesting!

steveatinfincia, 2020-12-07:
This is how things will work by default in OpenHD 2.1, all video is transported with RTP.

Timestamping should be happening already with rtph264pay, however there are some quirks in the way that stuff works depending on what the source element was.

closed 2023-01-31 by raphaelscholle