OpenHD/OpenHD #175: Is it possible somehow to connect AirPi to a wifi camera using similar hotspot as on RXPi

VU2ATX, 2019-10-11:
Hi,
My question is to developers of OpenHD. Is it possible somehow to connect AirPi to a wifi camera  using similar hotspot as on RXPi. I am not worried about the intereference at the moment.

I have an IP camera based on GK7102, this supports h264 and has two axis stepper motors control system too but unfortunately this camera does not have any LAN port and no datasheet GK7102 available (have googled a lot).

If this works then it can become quite interesting in low cost HD FPV. Some snaps are attached for reference.
Pls support, thanks in advance !
[image]
[image]
[image]
[image]

Beegad42, 2019-10-20:
It looks like the phone app uses a Wifi direct type protocol to initially communicate with the camera and configure Wifi to connect to a Wifi router. This could work for Open.HD hot spot but the next challenge is to instruct the camera to start streaming and so on. For go pro this is easier as someone has already hacked the communication between camera and router. So if you question was for a Gopro the answer would be yes, but no idea on this camera.

VU2ATX, 2019-10-26:
thanks, can you pls share the link for gopro work?

steveatinfincia, 2020-12-07:
An update to this, in OpenHD 2.1 you can set a particular wifi card (onboard or otherwise) as a wifi client to make it join a hotspot created by one of those cameras. It's not specifically intended for that but it would at least work at that level.

Then you'd have to treat it like an IP camera I think and put a section in `camera.conf` for that vehicle.

I'm not sure how well it would work but as soon as OpenHD 2.1 images are released you can try it.

VU2ATX, 2020-12-26:
Thanks a lot Stephen and sorry for the delay actually it got skipped, again sorry for that.
Would be in touch for more detail on Telegram.
-Narpat Singh Rana

raphaelscholle, 2023-01-31:
In general the wifi-hotspot is not limited so if somebody want's to work on this, it's duable, but not a good idea because of the bad quality internal wifi
(issue can be repoened if somebody wants to do that)

closed 2023-01-31 by raphaelscholle