OpenHD/OpenHD #176: TL-WN722N V2 "rtl8188" support

best4username, 2019-10-11:
I have several TL-WN722N V2 and try to offer support for it in Open.HD .
Started with OpenHD V2.0 RC1.1 image remove already exist driver (by blacklist module) , then search for a driver that support monitor mode found this from aircrack-ng team :
https://github.com/aircrack-ng/rtl8188eus
Clone it and try to compile first error that the compile procedure chose linux headers for (4.14.71+)
(there is two linux headers come with images , one is native raspian lite  and headers downloaded via Open.HD build scripts) , to avoid this i have removed the old linux headers now the compile take the correct kernel sources (4.14.71-v7+) and no errors when compile , but when i plug the wifi card this error show in dmesg :
  8188eu: version magic '4.14.71-v7+ SMP mod_unload modversions ARMv7 p2v8 ' should be '4.14.71-v7+ SMP mod_unload ARMv7

I dont know where this "modversions" come ???
How to tell make to not use this kernel symbol "modversions" (correct me if it not a kernel symbol )?
Also try to modprobe with --force-vermagic No luck ?
try to use dkms service with no luck....

steveatinfincia, 2020-01-31:
This should be no problem to integrate in the builder but I'm not sure if the rest of the system will actually support that card yet.

pilotnbr1, 2020-01-31:
This would be nice. Alot of v2 out there

Lekasl, 2022-04-17:
Hello. Alfa Awus036NHR V2 chip Realtek RTL8188RU work now ?
Thank you.

satyanvesi, 2022-06-03:
I dont have solution for your problem but you can read if you like.
I think that people not much talking about tplink wn722n V2/V3 (rtl8188eus). so here is my take.

Tx: pizero W with tplink wn722n V2(stock antenna) using pi tail OS
Rx: same as Tx

I am not using OpenHD.I am using befinitiv wifibroadcast.one reason for using pi tail is that it has Re4son kernel on its page it mentions that it has drivers built in for wn722n V2 which is true.
another reason is to use pi tail - i have tried every single os like in raspbian(buster,jessie etc non of them work i tested all) i receive very few packets you can say no packets at all,kali arm version i couldnt do ssh and also i dont have micro hdmi,pi tail works like a charm although it has driver still i compiled for aircrack-ng rtl8188eus driver.

I tested(stock antenna both side) in my at 12 foot distance i get 49 to 51dbm which doesnt seem good.there is no interfearance other than my mobile hotspot.

what do you think i am testing it wrong?
What is tx power of wn722n V2?(tx power is my main concern i dont trust value of iwconfig nor on txpower increase tutorial on web)
do i expect a good range about 1km?

satyanvesi, 2022-06-03:
Open.HD has rtl8188eus support now.

closed 2022-10-16 by raphaelscholle