OpenHD/OpenHD #180: High (10%) Video Packet Loss on RTL88xxau cards when UseMCS-'Y'

Beegad42, 2019-10-25:
Script tx_functions.sh fails to detect the setting UseMCS='Y' because the function datarate_to_wifi_settings (in file global_functions.sh) changes the value of UseMCS from 'Y' to '1'.

This causes the line "      	if [ "$CTS_PROTECTION" != "Y" ] && [ "$UseMCS" == "Y" ]; then
to fail and the wrong value for VIDEO_FRAMETYPE is passed to tx_rawsock causing packet loss.

A solution is to change this line to 
                      if [ "$CTS_PROTECTION" != "Y" ] && [ "$UseMCS" == "1" ]; then

But there are likely other bugs caused by the "hidden" change of value of UseMCS.

References:
https://github.com/HD-Fpv/Open.HD/blob/master/wifibroadcast-scripts/tx_functions.sh#L96
https://github.com/HD-Fpv/Open.HD/blob/master/wifibroadcast-scripts/global_functions.sh#L211

Beegad42, 2019-10-25:
This issue would apply to any image to which this patch has been applied (i.e. from about Sept 11)
https://github.com/HD-Fpv/Open.HD/commit/16a4e83cb6848c38826f5b75f2f1e88a41a6c4be#diff-ee5b51a03a0fb6dc48a9b82a49c75548

htcohio, 2019-10-25:
Ok,

Do you see anything or any possible problems related to Atheros cards (which do not use MCS rates and require CTS for RC?

Thanks!

htcohio, 2019-10-25:
BG, are you on open HD as a contributor? I'M just curious if you are able to actually submit a PR for this? 

Also, here are a couple links to past conversations about the whole RC and telemetry script related to open HD and Wi-Fi broadcast.

https://www.rcgroups.com/forums/showthread.php?2664393-EZ-WifiBroadcast-cheap-digital-HD-transmission-made-easy!/page82

https://github.com/rodizio1/EZ-WifiBroadcast/issues/170

Beegad42, 2019-10-28:
This issue is not related in any way to RC control or Atheros cards. It is related to 8812 cards only and the issue would occur regardless of whether RC is in use or not.

htcohio, 2019-10-28:
I see,

That is interesting because the last time I checked with those cards there was virtually no packet loss.

I'm going to get back into testing this week now that I'm done with my ground unit. 

Overall, do you see any possible changes or problems from the commits related to atheros cards as well?

I only ask because since around September there seems to be some reports of packet loss and latency with atheros cards as well.

I'm wondering if there could be some relation there or not.

P.S. atheros never uses mcs rates because they work best on Legacy rates.

Beegad42, 2019-10-29:
You probably tested before this was applied https://github.com/HD-Fpv/Open.HD/issues/180#issuecomment-546178891

I haven't looked in to Atheros cards, I only own 8812

pilotnbr1, 2019-11-26:
Is this still a problem?

htcohio, 2019-11-26:
No, er can close it

Justin Healey
HTC Communications LLC
O 440-523-1400
C 440-523-9578
www.htcohio.com

closed 2019-11-27 by pilotnbr1