OpenHD/OpenHD #181: Support for other SBC platforms

jonomakepeace, 2019-10-25:
For my use case RPi 0 air unit is probably the best bet. For other use cases, something like the RockPro64 might be an attractive option. As a ground unit, I think there are better hardware options than the RPi 3B+ that might improve performance on the ground side and also be more cost effective eg Rock64.

Has anyone tried to build the image for other hardware platforms such as Orange Pi, Pine64 etc? 

I would be happy to try this myself and port it to for example the Rock64.

What are the major obstacles here? I have a very naive view of the system so far (documentation on differences from standard images could be better), but keen to dig in if I can get an idea from someone more knowledgeable about the system on whether this is (easily) feasible. I have some experience with a angstrom+bitbake embedded Linux as well as Beaglebone, and a liiittle RPi.

htcohio, 2019-10-25:
Hi, some people are working on this but, it will probably be several months or years before anything is close to stable.

To put things in perspective, last year we worked to open up compatibility with the Raspberry Pi 3 B+ and 3A+ which for the most part was successful but to this day we're still working out some bugs.

Wifibroadcast/Openhd operates by taking advantage of Raspbian specific drivers like Raspivid and Hello_video...

Not only that, several functions involving the pipelines are handled in an efficient way with the Raspberry Pi specific system-on-chip architecture. 

Overall, your best bet for more processing power on the airside is to Simply strip down all the components of a RPI 2b+ 3b or 3B+... When this is done the board is only 19grams...  Just use a good set of wire cutters then cut off the USB, Ethernet, HDMI etc and you're good to go!!!

jonomakepeace, 2019-10-26:
Hi, thanks for your quick response and insight into this! 

I've done some digging through the code and image builder and I get what you mean - the RPi specific video drivers make things difficult! I've also done some research on the Pine64 and Orange Pi video capabilities and they don't seem to be there yet for this kind of application, so I guess we are stuck for now.

I tried to build the image for RPi 4 (3 B+ seems to have suffered a stock out, although I have managed to find one) and it seems that the main hurdle there would be 4.19 kernel support.

Going to close this issue as it seems irrelevant at this stage. Thanks!

closed 2019-10-26 by jonomakepeace