OpenHD/OpenHD #182: Both Air Pi  and Ground Pi don't work when 4G Dongle is pluged.

greentea878, 2019-10-26:
**Describe the bug**

I plugged a USB 4G Dongle to the air Pi (Pi 0 and Pi 3B)and/or ground pi (Pi 3B) before booting, both cannot work anymore.

If I plug the 4G Dong to ground Pi after system have started, with changing "sda" to "sda1" in the script, the ground pi can work.

The version I used is OpenHD V2.0 RC1.1.

Thanks

Greentea878

pilotnbr1, 2019-11-03:
Thanks for the info! I think you are the first to attempt this

mjc-506, 2019-11-30:
Ok. This is likely to be a modeswitch related issue:

When plugging in, the 4g dongle (and most usb modems) will advertise as a usb disk, to provide drivers/software for Windows machines. Once this is installed, a signal is sent to the dongle which changes to 'modem mode'and starts working.

Obviously, in Linux, these drivers don't work. Instead, there's a usb_modeswitch utility to send the magic signal to the 4g dongle to set it to modem mode, at which point the Linux drivers can take over and make it work. There is often a udev rule to run the usb_modeswitch utility automatically.

I suspect openhd fails to boot when the dongle is plugged due to fstab using /dev/sda style references (mucked up by the usb dongle disk). Changing to label or uuid references would solve this, and then usb_modeswitch can run once booted to set up the 4g dongle.

steveatinfincia, 2020-03-21:
Yea switching to uuid wouldn't be hard, and if someone can provide the USB VID/PID for the stick, I can add udev rules for it.

Running `lsusb` on the ground pi should do the trick.

pilotnbr1, 2020-03-21:
I can do this tomorrow for my huawei sticks

closed 2023-01-31 by raphaelscholle