OpenHD/OpenHD #185: Pi 4 Support is needed.

greentea878, 2019-11-01:
Pi 4 module is available in the market, Pi 4 support is needed to decrease the latency.

pilotnbr1, 2019-11-01:
How do you think it will decrease latency?

Summer88, 2019-12-18:
With 4GB of RAM would be nice for video files saved in tmp then transferred to USB at the end of mission.
Anyone has tried it?

steveatinfincia, 2020-02-19:
Update on the state of this issue:

I've added support for building pi4 images to the builder, and I've written a new video renderer that will work on the pi4, so it should only be a few weeks at the most before we have pi4 images to test.

FPSychotic, 2020-02-21:
Hi, I work with RPi4 on robotics, UGV mainly  but some UAV too. The point is that use ROS (a kind of framwork) over Ubuntu, it is possible run you software as a framwork, or run it over other OS instead a standalone and whole OS image?

steveatinfincia, 2020-03-21:
Pi4 support should be available later today, seems to be working well just needs to be merged into master and released.

For the moment only QOpenHD is available as the OSD on the ground side, the old OSD is written with OpenVG which is not directly supported on Pi4 (I don't recall if it's the new open source GPU drivers that don't support it, or the pi4 GPU itself). 

It's possible to use a compatibility layer for OpenVG on Pi4, but I don't know how well it would work or how much work would be required to go that route.

steveatinfincia, 2020-04-19:
Pi 4B support is now available in `nightly-2020-3-1` and later releases. A stable release will be provided in the next ~10 days.

closed 2020-04-19 by steveatinfincia

wx4cb, 2020-09-14:
I tried loading the latest release but it won't boot on the pi4. It appears that the dtbs are there, but there's no entry for the pi4 in the config.txt file.

```
[pi0]
start_file=zero_x.elf
fixup_file=zeroup_x.dat
[pi0w]
start_file=zero_x.elf
fixup_file=zeroup_x.dat

[pi1]
start_file=1to3b_x.elf
fixup_file=1to3bup_x.dat
[pi2]
start_file=1to3b_x.elf
fixup_file=1to3bup_x.dat
[pi3]
start_file=1to3b_x.elf
fixup_file=1to3bup_x.dat
[pi3+]
start_file=start_x.elf
fixup_file=fixup_x.dat
```