OpenHD/OpenHD #189: Pictures

Howie9600, 2019-11-07:
Is there a way to take pictures while flying?

Just a way to trigger a snap shot then transfer it to the groundpi?

htcohio, 2019-11-08:
Hey,

That is something that might be possible...  Many people have asked for video recording on the air side but there is a stability risk with that. 

However, grabbing a snapshot is much less cpu-intensive and could certainly be possible.

pilotnbr1, 2019-11-08:
Im sure you know there is already a screenshot function

htcohio, 2019-11-08:
I think he wants to trigger a "picture" like on DJI. 

Perhaps that could be done with the openhd Buttons

the-nicolas, 2020-08-31:
Thaz would be awesome. I even dont need to transfer the pictures to groundpi. I can copy them to some usb stick after landing or usi them scp.

And it would be good to have gps coordinates or also camera pitch etc. in the exif. And a way to trigger rhaz with mavlink from missionplanner. I just need to create camera tracks for photogrammetry.

Recording the video in full quality on airpi is not an option due to performance issues?

raphaelscholle, 2023-01-31:
Air-Recording ins added in 2.3

closed 2023-01-31 by raphaelscholle