OpenHD/OpenHD #190: Move OSD-params to main settings file

CopterGUI, 2019-11-10:
Hello Guys,

as I understand the GPIO switches are there to switch between config-profiles or models on the field. But right now I have to swap between different sdcards when using both iNav and Ardupilot models because of the difference in the OSD \ compass \ home-arrow. 
Because I have a closed GS it’s not ideal.

This issue would be solved by merging osdconfig.txt and openhd-settings-X.txt

or add a “master iNav Switch“ to openhd-settings-X.txt to solve this.

Greetings,
Sebastian

RespawnDespair, 2020-12-04:
For version 2.1 we will much improve the model switching options.
I will attach this issue to the 2.1 version board.

RespawnDespair, 2020-12-04:
Related to #385, closing this one as the other issue contains more details.

closed 2020-12-04 by RespawnDespair