OpenHD/OpenHD #195: SmartSync

Howie9600, 2019-11-13:
I am having issues with SmartSync.  I have the controller waiting for connection but the drone is not syncing.  Do I need to change something on the airpi?

Howie9600, 2019-11-21:
Any Help?  should it sync with default settings?

pilotnbr1, 2019-11-26:
You can trigger the sync with gpio or your transmitter...

Have you looked here-
https://github.com/HD-Fpv/Open.HD/wiki/Using-the-Open.HD-SmartSync-BootSystem

### And in openhd-settings-1.txt 
# >>>>>>>>>>[FEATURE: 2-WAY COMM APP]-- For Air/Ground settings with Open.HD APP.
# This enables or disables two-way communication between the open HD app and air/ground units.
# **Setting "2" disables AirPi remote settings (not needed anyways), 
# Primary benefit is to keep RPI zero CPU Usage down to a minimum with Atheros. Setting "1" Is helpful for quickly changing/testing
# settings on Air Side by immediately applying changes for comparison without the need for SmartSync. 
#Values "2"=Enables Ground Only, then uploads changes to air during SSync Boot (default) ; "1"=EnableAir/Ground Immediately save air changes through app...(Not recommended for regular usage)
RemoteSettingsEnabled=2

# >>>>>>>>>>[SmartSync Settings]--**SmartSync_StartupMode="0" means air/ground will boot quickly but can trigger SmartSync with CH2 movement or GPIO# 26 Cross Connected.
SmartSync_StartupMode=0
# RC Channel should default to middle (1500pwm position) to work properly
SmartSyncRC_Channel=2
# DO NOT CHANGE THE BELOW SMARTSYNC SETTINGS!
SmartSyncGround_Countdown=50
SmartSyncOFF_RC_Value=1700
SmartSyncStayON_RC_Value=1300
SmartSyncControlVia=joystick

Howie9600, 2019-11-26:
I can trigger the sync on the recieved just not on the drone.   Do I have to have something to trigger it on the drone?

Howie9600, 2019-11-27:
HI Luke

Sorry for being short with you.  I have a number of questions with your response.  So i have a RPI zero as my airpi and a RPI3b+ as my groundpi.  I have all the settings default and i force the smartsync by my RC channel.  The controller is then waiting for connection.  I power on my quad and nothing.  No syncing nothing.  The previous version this forced a sync and everything was good for multiple frames.  The wiki page seems out of date for the new release as it mentions smart sync 1 or 0 but the settings has 2.  Also in your response you mention SSYNC boot.  what is this  do you mean the boot when i have put it in to sync mode from the rc channel or is something else happening?

thanks

pilotnbr1, 2019-11-27:
It sounds to me like you are doing everything right. I was hoping Justin would comment.

In my opinion smart sync works well but has an interface problem. People constantly question what is happening... I think it needs improved messages telling users what has happened or not

Howie9600, 2019-11-28:
Hi Luke.  

Thanks for that.  If I am doing everything right and it's not syncing is it a bug?   I really need the ability to change the drone config remotely and noting I try is working.  I have tried a number of different rpis and nothing is changing.

Can you confirm that with default settings once I force SSync with my controller the drone is supposed to sync automatically?

RespawnDespair, 2019-11-28:
I believe it is turned off by default now.

```Remotesettings=1```

Should enable it (on both air and gnd in the config)

steveatinfincia, 2019-12-05:
It's still on by default, but there's a SmartSync bug in the 11/11 image.

TheNewJavaman, 2020-08-25:
Was this issue fixed?

Howie9600, 2020-08-25:
It was a issue with the hardware.  I swapped dongles and now it works

closed 2020-08-25 by Howie9600