OpenHD/OpenHD #196: RT2800usb driver not supporting monitor mode (kernal problem?)

notreallyaclue, 2019-11-19:
My wireless adaptor is recognised, its boots into Open.HD and gives a blank screen although waiting for a tx to be connected, but doesn't pick up any packets. When both the AWUS036NH and another supported adaptor are connected, the supported adaptor works fine, and the AWUS shows 100% dropped packets. 
The adaptor works with ezwifibroadcast fine. Upon inspection I found:
Both openhd and ez load version **2.3.0** of the driver from **rt2x00.serialmonkey.com**
However it seems the kernal is different, **4.9.35-v7** for ez and **4.14.71-v7** for OpenHD.

I have read issues about this adaptor and it was shown that one of the kali versions had an issue where it would drop packets if not destined for it: https://forums.kali.org/showthread.php?37575-rt2800usb

Unfortunately I'm not experienced enough with Linux to know where to start, I got as far as I could!

pilotnbr1, 2019-11-26:
Sounds like a fair amount of work to get this going... Please reference this page for supported adapters. https://github.com/HD-Fpv/Open.HD/wiki/Hardware-~-Supported-WiFi-adapters

notreallyaclue, 2019-11-28:
Okay I solved it,
Installing this package changes the driver used and fixes the regression issue with the above wifi card, allowing 2 watt cards to be used.

https://packages.debian.org/stretch/all/firmware-misc-nonfree/download

Mount the sd card in linux, put the file on the card. Boot openhd and go into terminal and install use dpkg. Reboot!

PS Not tested in flight but on the bench I get video and packets

RespawnDespair, 2019-11-28:
@pilotnbr1 might be worth it to include it in the builder? Presuming the adapter performs well in real life?

pilotnbr1, 2019-11-28:
@RespawnDespair sure! Happy to add a branch and set it up so it can be tested. Might take me a couple days to get around to it though

pilotnbr1, 2019-11-29:
Package added in builder new branch fix_rt2800usb...untested... Can someone test this branch with the appropriate card! Thanks

closed 2020-01-31 by steveatinfincia