OpenHD/OpenHD #198: SITL and UART testing

Summer88, 2019-11-28:
Hi, 
I've just followed the wiki https://github.com/HD-Fpv/Open.HD/wiki/RC-~-RC-with-Ardupilot,.

Then connected FTDI USB Serial Device converter now attached to ttyUSB0 to local PC and started SITL with:

simulator/ardupilot/APMrover2 && ../Tools/autotest/sim_vehicle.py -A "--uartC=uart:/dev/ttyUSB0" --console --map

I expected that data outputted by SITL would coming in the connected UART of AirPi, then see them on the OSD of the GroundPi.

I can see the TX AND RX ON the leds over the phisical cable blinking but no MAVLINK output on OSD of GroundPi.
Only video is received.

Instead if I start Qgroundcontrol on local PC I can see the data.

attached the configuration of both Air and Groundpi.

**Hardware:**
 - Air side: 
 -- Pi model: pizero
 -- Camera(s): picamera v2
 -- Wifi card(s): atheros 9271

 - Ground side
 -- Pi model: pi3
 -- Wifi card(s): atheros 9271
 -- Ground Control Station(s): openhd
 -- Hotspots - do you connect you GCS using wifi 
 - RC/telemetry uplink ???
 -- Do you use RC over Open.HD

I'm not much an expert of uart in past I've used picocom for debug, attached output of Airpi screen.

[openhd-settings-1.txt](https://github.com/HD-Fpv/Open.HD/files/3901307/openhd-settings-1.txt)
[image]

pilotnbr1, 2019-11-28:
Thanks for sharing all this! Its an interesting way to do dev work and try new things on the bench. I could see myself using this at times when it is not convenient to bring/power up a drone

Summer88, 2019-11-28:
I've got it working by changing speed of serial port:

# >>>>>>>>>>[USB OPENHD-RC SERIAL PORT/BAUDRATE]--Serial Port Settings between Air Pi & Flight Controller.
# Serial port and baudrate (19200 is minimum)
# Set this to "/dev/serial0" for Pi onboard UART serial port or "/dev/ttyUSB0" for FTDI adapter.
FC_RC_SERIALPORT=/dev/serial0
FC_RC_BAUDRATE=57600

# >>>>>>>>>>[OPENHD-FC SERIAL PORT/BAUDRATE]-- Air Pi and Flight Controller
# *Should be the same as FC_RC_BAUDRATE
FC_TELEMETRY_SERIALPORT=/dev/serial0
FC_TELEMETRY_BAUDRATE=57600

but seems that data are not updated constantly on the GroundPi.

closed 2019-12-02 by Summer88