OpenHD/OpenHD #200: GroundPi Date time from Airpi GPS?

Summer88, 2019-11-29:
This is not a bug, just a feature request, I'm a newbye but, I would like to get time synced for both Air and GroundPI.

Just ssh in Groundpi typed: date  return: Tue  9 Oct 14:01:25 UTC 2018

I know date time is in mavlink GPS messages:

> GPS times reported in MAVLink messages are in two parts: The GPS week number, and the number of milliseconds since the start of that week. Also, need to know when GPS weeks started (Jan 6, 1980).

Can please someone point me out to step needed to do this on Groundpi:
- read mavlink GPS date time: should I edit "wifibroadcast-base/tx_telemetry.c"?
- set system time: this should by bash script?
- add system time to OSD: should I edit "wifibroadcast-osd/render.c"?

Thanks, BR

pilotnbr1, 2019-11-29:
Mavlink.c is where you would set the message

Modify:
Telemetry.c
Telemetry.h

To view modify:
Render.h
Render.c

Find an osd item and just follow that same pattern through those files (like speed)..

I did some work with this some time ago to track telemetry latency. If you can wait a few days I think I still have it on my laptop at home and I can share it. It just goes off epoch time

Summer88, 2019-12-02:
Thank you for pointing me in right direction, I will try to get it up myself :-).
Thanks, BR

Summer88, 2020-01-02:
Hi, I'm triyng to add these data, but how can I test only osd part of program, when I start it complains:

> OSD: ERROR: Could not open /wifibroadcast_rx_status_0 - will try again ...

this is what have I done:
```
ionice -c 3 nice dos2unix -n /boot/osdconfig.txt /tmp/osdconfig.txt
cd wifibroadcast-osd/
make
/tmp/osd
```

pilotnbr1, 2020-01-02:
Take a look at screenshots.sh
It has some interesting lines that will give you hints on how to kill things and restart them. 

If it was me, for testing I would start a groundpi. Ssh in. Kill the stuff you dont want running (pretty much everything). The write modifications to the code via ssh. Make. Then restart the OSD if make was successful.

closed 2023-01-31 by raphaelscholle