OpenHD/OpenHD #201: V2.0 RC1.1 won't boot when camera is connected

maayan-scadafence, 2019-11-29:
Hey, i flashes version 2.0 RC1.1 to both pi 3b and pi zero w. The problem happens on both.
When i connected the pi camera module (original 5MP), the Pi didn't boot -just black screen. When i removed it, it booted successfuly
Worth to mention that EZ-wifi broadcast image did work and booted successfuly in the same configuration.

**Hardware:**
 - Air side
 -- Pi 3b & Pi zero w
 -- pi camera module 5MP
  Thanks!

pilotnbr1, 2019-11-29:
No one else is reporting this... power problem maybe? Something corrupted? Bad cam?

maayan-scadafence, 2019-11-30:
I doubt it will be in the log because the OS Didnt even start to boot, ill check it though.
Also ill try again in the ez-wifibroadcast image, to recheck that the camera is ok.

mjc-506, 2019-11-30:
Does the airpi display anything once booted?

maayan-scadafence, 2019-11-30:
Doesnt display anything, black screen

maayan-scadafence, 2019-11-30:
It's probably faulty camera module board, bcz now i flahsed ez-wifibroadcast and still didnt boot

steveatinfincia, 2019-12-05:
OpenHD turns off the console entirely until a specific part of the scripts run, making it look like nothing is happening.

The file is `cmdline.txt` on the boot partition (which can be edited on any PC), remove `fbcon=map:2` and `quiet` and then you should see the kernel boot log and the initial Open.HD scripts running, which should show an error of some sort. 

You might want to take a video of the screen as the screen may keep "scrolling" past the error once it shows up.

maayan-scadafence, 2019-12-06:
@steveatinfincia that's great to know, ill try It thanks!

lasselars, 2020-01-26:
It seems I have the same problem but on rc 1.2.
Installed version 2.0 rc1.2 on Raspberry Pi W and if I connect camera it won't boot up.
(Open.HD_V2.0_RC1.2-2019-11-24_RSSI-FIX.img)
The rpi and camera is working. Tested with raspbian and raspivid.

Yes21, 2020-01-26:
Which wifi board are you using ? Is it powered by the Pi 0 W ?
Each time my Pi0 didn't boot, it was because it wasn't able to power the wifi board ...

lasselars, 2020-01-26:
I just tested with a Raspberry Pi3+ and another camera without the Alfa awus adapter and still the same. Does not boot with camera connected and the rc 1.2 image. Booting on Raspbian okay.
I have now tried two different cameras on the Rpi Zero W and a third camera on the Rpi3+ all with the same result. No booting.

My initial test on Zero was with a powered usb hub connected to give good power to the Alfa awus card.

If it had booted and then stopped it would have been easier to troubleshoot. 
I am new to the OpenHD so I have not yet read so much about it's configuring.
(Rpi and cameras I have since they came out)

maayan-scadafence, 2020-01-27:
My problem was with the camera board. I booted with ez-wifibroadcast and it didnt boot either, and the error was a hardware fault on the camera.
Did you try changing the camera _board_ itself, not only the optical sensor ?

lasselars, 2020-01-27:
I tried three different camera boards. One of them is a Raspberry Pi official board and the others might be some 3:rd party board. All of them are working on their respective RPi with Raspbian. 
I also swapped them around and they worked. 
I then tried OpenHD one zero and 3+ withl all of those three boards without success.

I think I have to read more and find out if I can do some settings to stop probing for camera board and then find out more after OpenHD has booted. If I remove the camera cable the zero and 3+ boot.
Ideas are welcomed.

maciek252, 2020-01-28:
I have exactly the same problem. Two cameras, one V2.1 and another NoIR V2. If a cam is connected, no boot, no video, Ground side reports nothing. Without a cam, it boots to Ground. The cams are working with Raspbian. I'm new to Open.HD and don't know yet how it should work. But something seems to be wrong... Open.HD version "Latest release", i.e. Open.HD_V2.0_RC1.2-2019-11-24_RSSI-FIX.img

steveatinfincia, 2020-01-31:
When you say it doesn't boot are you referring to lack of output on a screen connected to the air pi? 

By default it doesn't show anything on the screen at all, even when things are working. I'm going to change that tonight though as it's making it difficult for people to see what's going on and difficult for us to help :) 

In the meantime, what wifi cards are you using on ground and air?

maciek252, 2020-01-31:
I mean the lack of any reaction at all. No image received from the GroundPi. I haven't yet reached the OpenHD running, so I don't know how it should be.

EZ-wifibroadcasts emits some messages when the AirPi is alive and transmitting, it indeed helps a lot.

I'm using 2x TL-WN722N.

lasselars, 2020-01-31:
> When you say it doesn't boot are you referring to lack of output on a screen connected to the air pi?
>
> By default it doesn't show anything on the screen at all, even when things are working. I'm [...]
Little do I know
You are correct. I assumed it is not booting, when camera is attached, since I don't see anything on the screen at all.

Info. 
I have the possibility to test variants if I just understand what and how to test. I have most of the Raspberry Pi models since the start except for the A-series and the RPi4 and I have also three different cameras to try. One is "original" and two are something from AliExpress or DealXtreme.

WiFi cards
I have purchased three wifi cards, two brands (usb attached) with, according to Linux it's identifying itself as: 
Ralink Technology, Corp. RT2870/RT3070 Wireless Adapter.
The cards are:
(N9000 Wireless Adapter) Alfa Network. 2W AWUS036NH 802.11b/g/n Long-Range USB Adapter  
(KuWFi Alfa USB Wifi Adapter) AWUS036NH Ralink3070L

careyer, 2020-01-31:
2x TL-WN722N of which hardware revision? V1.1 or newer?

Von meinem iPhone gesendet

> Am 31.01.2020 um 10:37 schrieb Maciej Szreter <notifications@github.com>:
>
> 2x TL-WN722N

maciek252, 2020-01-31:
2x TL-WN722N  Ver. 1.10 - labels on the devices and boxes as well.

jcausemann, 2020-01-31:
some days ago i got thid tip on telegram:
OpenHD turns off the console entirely on the air side (in other words, if a camera is detected), until a specific part of the scripts run, making it look like nothing is happening.
The file is cmdline.txt on the boot partition (which can be edited on any PC), remove fbcon=map:2 and quiet and then you should see the kernel boot log and the initial 
Open.HD scripts running, which should show an error of some sort.

careyer, 2020-01-31:
You can also set DEBUG=Y in the settings-openhd-1.txt file

steveatinfincia, 2020-01-31:
There will be a new nightly image today that will always show the boot process on ground/air regardless of whether a camera is connected.

lasselars, 2020-01-31:
Thank you. Yes, it is working. I did was what explained in two posts. 
DEBUG=Y and removed fbcon=map:2 and quiet.
Now I need to do some settings. Every second the image gets "pixelated". Even (or because?) I have the two network cards close to each other.
~5Mbit, 126 dBm    7/12

steveatinfincia, 2020-01-31:
The [nightly-2020-1-31](https://github.com/OpenHD/Open.HD/releases/tag/nightly-2020-1-31) release from an hour ago has the console enabled by default now, so those of you having trouble please try it and let us know what happens.

maciek252, 2020-02-01:
Thanks! I'll report but I need to get more SD cards as I don't want to lose my already working EZ-wifibroadcast setup.

maciek252, 2020-02-07:
I've checked and now it works, the video is being transmitted from Air to Ground. Also the messages from Air appear on its HDMI output. Thanks!

maciek252, 2020-02-09:
Unfortunately it worked fine only once, then stopped again. At least Air displays messages, it hangs on this: "Starting Hold until boot process finishes up". Broken file system?

steveatinfincia, 2020-04-19:
Sorry for the delay getting back to you, it does sound like an SD card issue in that case.

In the past few releases we have been able to see the boot process on the air side if a screen is connected and it seems to work well now. 

I'll close this for now and if anyone is still having an issue we can reopen it.

closed 2020-04-19 by steveatinfincia

timbuktu031, 2021-02-18:
I am same issu.