OpenHD/OpenHD #202: TxPowerAir changes get applied only after a second reboot with Atheros card

estogantsev, 2019-12-01:
**Describe the bug**
openhd-settings-1.txt TxPowerAir changes get applied only after a second reboot of the AirPi with Atheros card.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect the AirPi wifi adapter to the power meter
2. update TxPowerAir in openhd-settings-1.txt
3. put the microsd card into the air pi
4. boot up, see the power reading -> not changed
5. unplug and plug back in -> now the power corresponds to what was set

**Expected behavior**
The change should get applied on the first boot

**Configuration and debug logs**
the only change from defaults in openhd-settings-1.txt is the TxPowerAir value
the only difference in dmesg output happens on the first boot after TxPowerAir is updated:
...
[   33.291017] ATH: TX Power set: 42
[   33.499015] ATH: TX Power set: 42
[   33.996018] ATH: TX Power set: 42
[   34.499352] device 0011701fed20 entered promiscuous mode
[   39.045042] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)             <- on first boot only
[   39.270965] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)             <- on first boot only
[   40.522077] device 0011701fed20 left promiscuous mode
....

looks like the new power setting is copied somewhere onto the root partition but doesn't get applied on the first boot.

OpenHD image: Open.HD_V2,0_RC1,1-2019-09-09.img

**Hardware:**
 - Air side 
 -- Pi model: Pi Zero W
 -- Camera: Pi Camera V1
 -- Wifi card: "dogshit" blue Atheros card from aliexpress

 - Ground side is powered off / same effect if it is on

htcohio, 2019-12-29:
Hello,

Sorry for the delay, I totally overlooked this on accident. 

Yes, That is a known issue that has to do with the Kernel driver & Firmware withing the Atheros hardware. 

It is one of the "handicaps" of the project... Ath9k_htc utilizes iee80211 but newer RealTek RTL8812AU for example operates differently and, IS able to adjust "txpower on the fly".

Ultimately, it comes down to what changes can be made or not made with soft reset or live changes on various wireless cards. 

It is annoying I know but, unfortunately this has been explored many times with many headaches :(.

P.S. Ironically, some things like Bandwidth live switching is supported.  e.g 5mhz/10mhz/20mhz on the fly. Have you tried this at all?  Those rates have really great penetration and range (At the expense of limiting throughput to 1/2 or 1/4 of the 20mhz bandwidth.

steveatinfincia, 2020-01-01:
Sounds like this would work as expected if we just grab the txpower setting from openhd-settings-1.txt during boot and write it into the module parameter file before the module loads.

Shouldn't be too hard to fix.

estogantsev, 2020-01-02:
As I understand, txpower is copied from openhd-settings-1.txt after SmartSync is executed, so another possible solution is to trigger a restart if txpower was changed (either in openhd-settings-1.txt or via SmartSync). But in this case we need bootloop protection if copying fails.

htcohio, 2020-01-02:
> As I understand, txpower is copied from openhd-settings-1.txt after SmartSync is executed, so another possible solution is to trigger a restart if txpower was changed (either in openhd-settings-1.txt or via SmartSync). But in this case we need bootloop protection if copying fa

Hey,

yeah that's exactly correct.  I believe there is some logic built into the script to compare the before and after whereas if it changes compared to before and after ssync it might actually already do what you are suggesting.
*I believe Vitaliy/User1321 (Who wrote Ssync script). Struggled trying to figure that out. 

If 8812 cards are used, txpower changes during ssync do not cause any problems but, with atheros they do. 

I just want to remind you that SSync was originally structured so that the air side never gets hung up waiting during boot so if there's a unexpected restart midair or something it will always load with the previous settings. 

With that said, I'm not sure about you guys but, if there's a parameter I want to change, I usually just power the ground side first, Make changes, and power air.... Ssync runs and all is usually fine. 

Stephen, it seems like there should be some way to make the bug a little more logical or easier to deal with like you suggested.  Are you thinking that we should have some way to transfer some of the settings for the critical or flaky things like driver settings & txpower while both air and ground are powered up? 

If so, Then we could just clarify to users if they want to change TX power they can do it manually in the settings file as they always have, then have a note or disclosure that air should be powered on when changing txpower? 

In both scenarios, it's still kind of an additional step.

Perhaps we could display the current txpower value on the air side within the app or OSD? (its just 1-63)... so if somebody changes TX power and there's a problem it will clearly indicate it has not been applied yet somewhere on screen. 

Thoughts?

htcohio, 2020-01-02:
there are a few references to the stuff online as I'm sure you can imagine lol.

I have not read all of this yet but, you might find it helpful.

Also, there are boat loads of good information in this QCA GitHub repo if you browse through the open and closed issues. 

https://github.com/qca/open-ath9k-htc-firmware/issues/72

htcohio, 2020-01-02:
sorry accidentally closed this issue didn't mean to do that...

HD-Fpv, 2020-01-03:
here's an interesting link about the lock up and soft reset related items to the driver

http://lkml.iu.edu/hypermail/linux/kernel/1502.1/00446.html

closed 2023-01-31 by raphaelscholle