OpenHD/OpenHD #203: 40meg download video file to usb stick

Flierman1945, 2019-12-05:
After a flight, disconnecting Airside, inserting usb stick into Pi3B+ and downloading the video file, only up to 40meg is saved.  You can have a 20 minute flight and expext the file to be large, but it seems to have an upper limit at this present time.

Flierman1945, 2019-12-05:
After a flight, disconnecting Airside, inserting usb stick into Pi3B+ and downloading the video file, only up to 40meg is saved. You can have a 20 minute flight and expext the file to be large, but it seems to have an upper limit at this present time.

estogantsev, 2019-12-06:
what's your datarate (and video bitrate if not auto) set to? on the pi3b+ the tmpfs used for video and telemetry storage is about 250mb

Flierman1945, 2019-12-06:
Well ,I fixed it myself.Looking at the usb stick, I could see Two partitions.  The one where video rssi etc is stored only had a 43meg capacity. So in disk manager  "win10", I deleted the Two partitions and combined the unallocated space into One simple volume.  Hey Presto, job done.  Now able to record large enough video files for all my needs.  I would guess the max video file size will be somewhat less than the ram of the gnd Pi.

closed 2019-12-06 by Flierman1945