OpenHD/OpenHD #220: Ralink RT3572 and RT3070 RSSI

pilotnbr1, 2019-12-30:
Ralink RT3572 and RT3070 have been showing an rssi locked at -30 since about sept.

jcausemann, 2020-01-24:
Same with Ralink RA5572N 
Where can I download this older image, please?

jcausemann, 2020-01-24:
[image]

htcohio, 2020-02-03:
> 
> 
> [image]

Hey there,

Yeah, there should just be 1 or 2 files needed to get things back.  Since the Openhd System was optomized for the readily available and powerful RTL8812AU and Atheros AR9271 chipsets, some tweaks were needed for the newer MIMO/STBC rssi radiotap stuff which, of course, caused problems with older Ralink chipsets. 

Will either 

1- get back to you quickly with custom updated image with legacy RSSI settings.
2- Release a new image with the legacy RSSI repaired on the latest updates.
3- Share a couple rssi files you can simply swap out manually.

jcausemann, 2020-02-03:
Hey
Would be very nice if i could use newest images with my actual little racer.
[image]
Next build will be with AC56
Thanks

RespawnDespair, 2020-12-04:
@jcausemann Sorry for the incredibly long delay, I'm, a big fan of your drones!
I would hope this issue is resolved on the latest Open.HD release? 
Could you please verify?

I will leave the ticket open until we have verification it is actually fixed.

domenic64, 2020-12-04:
Test OpenHD v 2.0.6   Rt3070 rssi -126

RespawnDespair, 2020-12-04:
Thank you for testing, I will add it as a bug to the backlog for 2.0.

steveatinfincia, 2020-12-09:
This is fixed now, will be in 2.0.7.

closed 2020-12-09 by steveatinfincia

steveatinfincia, 2020-12-09:
RSSI working with Ralink cards (and a usb thermal camera, which wasn't working before on ralink either, fixed at the same time)

[image]