OpenHD/OpenHD #232: Atheros AR9271 (Ath9k_htc Optomizations)

htcohio, 2020-01-05:
*** Will add more details, Just posting a few starter notes related to some minor adjustments and improvements to the atheros driver to improve stability. 

1. Eliminate the 2nd antenna in firmware as it brings no benefit and only causes problems. 

2. TXpower values are inconsistent with our power patches.

3. Verify gains and TX power offsets are correctly set depending on closed-loop/open loop hardware Power Amplifiers. 

4. Investigate possible Packet aggregation and RC Uplink scheduling. 

5. Instabilities with >2 Ar9271 cards. Wont work with 3 Atheros + Ethernet.

webbbn, 2020-03-17:
Patches (to the patches) to support on-the-fly tx power changes for both ath9k and rtl8812 are here: https://github.com/webbbn/Open.HD-Kernel-Builder/commit/e03aee47742066a23403fca51a5f42950f7fa45b

didadidadida, 2021-05-07:
Indeed, when I use ar9271 network card and WiFi to inject data in monitoring mode, there will be about 10% packet loss. I don't know whether it is the problem of transmitting or receiving, although the FEC error correction algorithm of 8 + 4 is used. I use the original befinitiv's WiFi broadcast project to play image transmission directly. The packet loss rate is about 10%, and the reception is often green screen.

didadidadida, 2021-05-07:
What's the difference between monitoring mode and management mode? In the management mode, when the network card is connected to the router, the UDP packet loss rate is almost 1% in iperf3 test. UDP is not retransmitted. The firmware algorithm of network card should also use FEC (or driver layer)

Consti10, 2023-02-01:
closing, in general, rtl8812au and 5.8G outperforms 2.4G and AR9271 in pretty much all aspects.

closed 2023-02-01 by Consti10