OpenHD/OpenHD #234: Switch from monitor mode to IP in LAN

Summer88, 2020-01-07:
I need to disable wifi monitor mode and force traffic to LAN, 

I've added the setup of the IP on:
Open.HD/RemoteSettings/Ground/helper/ConfigureNics.sh

changed to consider only "eth0":
`NICS=`ls /sys/class/net/ | nice grep -v intwifi0 | nice grep -v lo | nice grep -v usb | nice grep -v intwifi | nice grep -v relay | nice grep -v wifihotspot`

Now  I get an error that says:
` Unknow encapsulation on eth0!  check if monitor mode is supported and enabled `

So searching I've found it in `Open.HD/wifibroadcast-base/rssirx.c` whre it says:

`ERROR: unknown encapsulation on %s! check if monitor mode is supported and enabled `

If Ground has address 192.168.1.1 and Air 192.168.1.2 how can I replace the open_and_configure_interface function to allow comunication via IP?

sni-pi, 2020-01-30:
Just reading about it on svpcom / wifibroadcast is there a way to enable **IPoverWB**  with OpenHD?

Summer88, 2020-01-31:
I'm triyng to send video over IP from airpi with:

`raspivid -o udp://192.168.10.10:9000 -w 1280 -h 800 -fps 30 -b 100000 -g 5 -t 0 -cd H264 -n -fl -ih -pf high -if both -exports -mm average -awb horizon `

then need to show the video on ground:
`
@wbc(boot:rw,root:rw):/home/pi# raspivid -l -o udp//127.0.0.1:9000`
will result in :
> mmal: Cannot read camera info, keeping the defaults for OV5647
> mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
> mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
> mmal: Failed to create camera component
> mmal: main: Failed to create camera component
> mmal: Camera is not detected. Please check carefully the camera module is installed correctly
> 

This works instead :
air:
` raspivid -w 1280 -h 800 -fps 30 -b 100000 -g 5 -t 0 -cd H264 -n -fl -ih -pf high -if both -ex sports -mm average -awb horizon -o - | gst-launch-1.0 -v fdsrc ! h264parse ! rtph264pay config-interval=10 pt=96 -o udp://192.168.10.10:9000
`
ground:
`gst-launch-1.0 udpsrc port=9000 ! tee name=t ! queue ! udpsink host=127.0.0.1 port=9000 t. ! "application/x-rtp,media=video" ! rtph264depay ! video/x-h264, stream-format="byte-stream" ! filesink location=/dev/stdout | /opt/vc/src/hello_pi/hello_video/hello_video.bin.48-mm > /dev/null 2>&1 &
`

Anyone has advices, please?

htcohio, 2020-02-02:
Sorry for the Dealy....

Checking on this.

FYI, if you want to console into the pi, you have to set DEBUG=Y in the openhd-settings-1.txt file.

If you have a Pi3b+ it should auto configure internal hotspot where you can also SSH into 192.168.2.1.

**Will get back to you on the other questions.

Summer88, 2020-02-03:
Hi,  thanks for your reply, about hotspot I'm not triyng to access data on groundpi from another device.

I'm triyng to make LAN conenction for both air and groundpi.
I've an external wireless Access Point, both of them are connected to this, assigned static ip addresses and are available from ssh on another lan: 192.168.10.x/24.

Summer88, 2020-02-14:
Just an advice: should replacing of mode be enought to make it work in the same Layer 2 LAN?

mode=
0 : monitor
1: promiscous

`interface->ppcap = pcap_open_live(name, 120, mode, -1, szErrbuf);`

RespawnDespair, 2020-12-04:
I'm not exactly sure of the use case here, but as the issue is quite old I am going to close it.
Lots of things have changed since this issue was created, so I hope the use case you were trying to support has been covered already. If not, feel free to reopen the issue.

closed 2020-12-04 by RespawnDespair