OpenHD/OpenHD #235: Integrate Qczek LRS LoRa with OpenHD as backup RC control/telemetry system

krzysztofkuczek, 2020-01-16:
Hi,
I wonder if I could somehow integrate my free available LoRa based LRS solution with open HD.
Thinking about using it as backup RC controll, and telemetry.
I can implement custom input/output protocols for both, just need to agree what would be possible due to limited link capacity.
Regards
Kris

pilotnbr1, 2020-01-17:
Hi Kris, Thanks for the interest! Always interested in adding more features. One question though, is all of your code open source? Trying to keep things transparent and easy to modify for us and others

krzysztofkuczek, 2020-01-17:
Hi,
Sorry but it's not open sourced. It's free available for everybody, and configuration protocol is well described, and no any restriction of usage. 
I can introduce new protocol, which will be well defined to integrate  with OpenHD, but I'm not going to open the source code.

pilotnbr1, 2020-01-24:
> Hi,
> Sorry but it's not open sourced. It's free available for everybody, and configuration protocol is well described, and no any restriction of usage.
> I can introduce new protocol, which will be well defined to integrate with OpenHD, but I'm not going to open the source code.

Talking to some of the guys that do long range flying, they have documented that the rc link of openhd is more robust than LoRa (flights up to 75km, only limited by UAV battery). Additionally, some have tried 4xx and 9xx links and found that it actually disrupts the video signal and reduces range. So there seems to be little interest in integrating a LoRa solution...

steveatinfincia, 2020-06-11:
When you say you don't want to open the source code do you mean the code running in the LRS device itself?

RespawnDespair, 2020-12-04:
To keep the boards clean I will close the issue for now.
If an interest in this still exists, we can discuss it further and add it to the roadmap, if so, please reopen.

closed 2020-12-04 by RespawnDespair

VU2ATX, 2021-03-03:
> Hi,
> I wonder if I could somehow integrate my free available LoRa based LRS solution with open HD.
> Thinking about using it as backup RC controll, and telemetry.
> I can implement custom input/output [...]

Hi, can I upload waypoints and change parameters on the fly in Ardupilot using QCZEK LRS?