OpenHD/OpenHD #236: Release an OpenHD version executable via command line

massimilianonball, 2020-01-17:
Hi all, mine it's only a request. 
I have several script already running on a raspberry on my drone. 
OpenHD project is amazing but it woluld perfect it there was a "core" version, usable via command line as a simple terminal command, rather then only the image for the sd card.
So for example something like:
```
on TX side: raspivid <options> – | ./openhdCORE_tx <options>
on RX side: ./openhdCORE_rx <options> | gst-launch-1.0 <options>
```
Something similar to the Vasily Evseenko's [WiFiBroadcast](https://github.com/svpcom/wifibroadcast/wiki) project but with all the extra functions that OpenHD have. 

Many thanks!!

htcohio, 2020-02-02:
Hey, So are you just trying to console into the air or ground side if needed?

FYI, almost all common settings are inside a massive Openhdsettings-1.txt file on the SD Card which will soon be unnecessary because we will have presets inside the new QOpenHD Android/Windows/Linux App as selectable presets based on hardware specific setups. + There are many relevant notes within there.

With that said, in order to SSH into AIR PI with an Ethernet cable or GROUND PI with Hotspot/Ethernet just set Debug=Y

From there, the pi pulls a 192.168.0.xx  (Or you can SSH into the 192.168.2.1 hotspot on ground pi.

massimilianonball, 2020-02-03:
Hi and many thanks for your answer!! 
As i mention in my previous post i already have several python and bash script for control the camera, the 4g connection, 2 video feed (one for the pilot of the drone and one from the mirrorless camera). 
I'd like to add the wifibroadcast streaming to my system for an automatic switch with the 4g in case of necessity.
All those scripts run on top of a raspbian lite on the raspberry, i think that run all of them in a pre built image like openHD will give a lot of problems.
A command line version of openHD will be for sure more easy to integreate in an already running system like mine. 
For now i will enable the ssh access as you suggest and i will try to add my scripts to the openHD system, i'm curiose to see what will happen, many thanks!!

RespawnDespair, 2020-12-04:
I really hope you have succeeded in this.
If you need any help with this, please reopen or create a new issue, closing this one.

closed 2020-12-04 by RespawnDespair

rspandolfi, 2024-12-23:
> Hi all, mine it's only a request. I have several script already running on a raspberry on my drone. OpenHD project is amazing but it would be perfect if there was a "core" version, usable via command line as a simple terminal command, rather then only the image for the sd card. So for example something like:
> 
> ```
> on TX side: raspivid <options> – | ./openhdCORE_tx <options>
> on RX side: ./openhdCORE_rx <options> | gst-launch-1.0 <options>
> ```
> 
> Something similar to the Vasily Evseenko's [WiFiBroadcast](https://github.com/svpcom/wifibroadcast/wiki) project but with all the extra functions that OpenHD have.
> 
> Many thanks!!

Nearly 5 years later I have a similar question. I need to run YOLO on the companion computer,