OpenHD/OpenHD #238: TC358743 HDMI-in board, video gets laggy

GlidLov, 2020-01-21:
Hi all,

TC358743 based board connected to a RPi3b+ or 3A+ (i have both) and Asus AC56 on both ends.
a sony A7R camera is connected to the HDMI-In board. All defaults apart freq chan which is set to 5805mhz. installed is the 12-1-2020 nightly

problem: video get laggy (2-3 secs) shortly after boot, exactly when bitrate indicator raises above 10mbit, then Camera icon start flashing in red. If i leave the camera untouched so the image can be more static, bitrate lowers and lag became low again. Everything is quite random, however pretty easy to reproduce.
I think it's something in the TC358743 driver (there's a warning on airunit console which says TC358743  driver is in alpha stage) or bitrate settings/management.

GlidLov, 2020-01-24:
small update: while playing with settings (bandwidth, basically) with a raspicam "standard" setup, i managed to have the same problem of laggy video. I enabled bandswitcher option, lowered band usage to 55% and put a fixed bandwitdh value instead of auto, i tried 4500000, 5500000, and 10500000, both had issue and got laggy. I resolved putting back to auto. Maybe this could help to locate problem

steveatinfincia, 2020-01-25:
This is likely related to issue #240, what is the framerate set to?

GlidLov, 2020-01-25:
Damn, it's 30 fps! I will try to halve bitrate percentage as soon as i'll try it

htcohio, 2020-02-02:
> 
> 
> Damn, it's 30 fps! I will try to halve bitrate percentage as soon as i'll try it

Hey, Yeah that is a sill little bug that has something to do with the BCM Chip and like how the Compute Module has 2 x CSI lanes per camera instead of just one in all of the standard Pi models.

Because the Raspberry Pi foundation does not "Officially" Support the TC358743 cards they don't really have it as a priority. 

We should have a fix implemented in the next release that auto reduces the txt file Bitrate% by 1/2 if 30fps is set. 

That should eliminate a lot of the confusion about this.

Also, one more note, in the settings for raspivid "EXTRAPARAMS" I added a shorter recommended extraparams list to Uncomment and use instead of default. (It just disables the Auto exposure/white balance etc.. that all Action cameras do, and do better anyways..... (Nothing good comes from processing the same exposure/white balance twice for no reason). 

Plz advise if we can close.

Cheers

htcohio, 2020-02-08:
Can we close this?

GlidLov, 2020-02-08:
> Can we close this?

Yes!

GlidLov, 2020-02-19:
little update:
with the last nigthly 14/02/2020,  setting FPS to 25 do *not* seems to improve things

htcohio, 2020-02-19:
25FPS is the cam setting? 

That means PAL format.

If there is NTSC format, in camera 30fps is better because, there is no structure setup to handle 25fps.

steveatinfincia, 2020-02-19:
I can add another special case bitrate reduction for 25fps, but does 25fps even work?

htcohio, 2020-02-19:
technically it should work with any frame rate but, problems can occur if
there strange combo between HDMI Actual fps and Raspivid Fixed Framerate.

Also, Hello_Video is tied to the Raspivid FPS.

So, If HDMI out is 25 fps

And, Raspivid is 30fps

And Hello_Video 30fps is trying to maintain jitter-free video, it will
likely cause problems at a couple locations.

Generally, if HDMI's FPS is the same or Higher than Raspivid, all is fine.

*I believe with NTSC HDMI out is usually only 30fps or 60fps...

Justin Healey
President
HTC Communications LLC
O 440-523-1400
C 440-503-5803
www.htcohio.com

GlidLov, 2020-02-20:
what i did is simply to set 25 fps into the openhd config file, no other changes. I don't remember how the camera is set (i wired a sony RX100 mk4 but it's very similar to the A7R in this). I will check

RespawnDespair, 2020-12-04:
@GlidLov is this issue still occurring in the latest Open.HD release? Sorry for not checking back sooner.

GlidLov, 2020-12-07:
hi all, thanks for keeping this alive. I don't have the gear to test it anymore. I will rebuild it then have a try, but no time lead ATM

closed 2022-10-16 by raphaelscholle