OpenHD/OpenHD #240: Adjust HDMI input bitrate automatically

steveatinfincia, 2020-01-25:
The HDMI input adapter causes an odd bug in the way bitrate is calculated, according to those who have one.

When the frame rate is set to 30fps, the bitrate percentage setting in Open.HD must be set to half to avoid bitrate spiking beyond the WFB data rate.

A short term fix would be to detect that the HDMI input adapter is connected (if the module is loaded), and then if FPS is set to 30, divide the bitrate percentage by half.

Long term we should figure out what actually causes this.

careyer, 2020-01-26:
Exactly the same thing holds true for the InnoMaker Veye IMX290 camera btw. It would be much appreciated if this can also be addressed with this fix (or with a seperate one - whatever is considered cleaner code wise)

Thanks! 
Cheers

steveatinfincia, 2020-01-27:
Does the problem with imx290 bitrate also happen with 30fps only, or is the camera always 30fps anyway?

careyer, 2020-01-27:
The InnoMaker Veye IMX290 is always 1080p @ 30fps. There is no other resolution or framerate to choose ;-)

commit c2b93c16 (2020-01-30): Cut bitrate in half for 30fps HDMI in and imx290

Closes #240
closed 2020-01-30 by steveatinfincia