OpenHD/OpenHD #260: Wireless card cannot switch from monitor mode

Summer88, 2020-02-07:
I'm triyng 2 wireless USB that present different behavior:
-(A) Bus 001 Device 015: ID 0cf3:9271 Atheros Communications, Inc. AR9271 802.11n
-(B) Bus 001 Device 016: ID 2001:3c1b D-Link Corp. DWA-127 Wireless N 150 High-Gain Adapter(rev.A1) [Ralink RT3070]

When I plug (A) in dmesg a lot of those message appear (is this supposed to be correct?):
```
[  865.043531] ATH: TX Power set: 58
[  865.175527] ATH: TX Power set: 58
[  865.307586] ATH: TX Power set: 58
[  865.439607] ATH: TX Power set: 58
[  865.571534] ATH: TX Power set: 58
[  865.703547] ATH: TX Power set: 58
[  865.835525] ATH: TX Power set: 58
[  865.967528] ATH: TX Power set: 58
[  866.099530] ATH: TX Power set: 58
[  866.231524] ATH: TX Power set: 58
[  866.364554] ATH: TX Power set: 58
[  866.496546] ATH: TX Power set: 58
[  866.673662] usb 1-1.1.2: USB disconnect, device number 13
[  868.736434] ath: phy8: Could not kill baseband RX
[  868.737432] ATH: TX Power set: 58
[  868.740690] ATH: TX Power set: 58
[  868.848939] ath: phy8: RX failed to go idle in 10 ms RXSM=0xc0fbb4
[  868.954660] ath: phy8: RX failed to go idle in 10 ms RXSM=0xc0fbb4
[  869.059631] ath: phy8: RX failed to go idle in 10 ms RXSM=0xc0fbb4
[  869.083595] usb 1-1.1.2: ath9k_htc: USB layer deinitialized
[  872.537086] Voltage normalised (0x00000000)
[  873.769080] usb 1-1.1.2: new high-speed USB device number 14 using dwc_otg
[  873.877163] usb 1-1.1.2: New USB device found, idVendor=2001, idProduct=3c1b
[  873.877176] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  873.877184] usb 1-1.1.2: Product: 11n Adapter
[  873.877191] usb 1-1.1.2: Manufacturer: Ralink
[  873.877199] usb 1-1.1.2: SerialNumber: 1.0
[  873.955044] usb 1-1.1.2: reset high-speed USB device number 14 using dwc_otg
[  874.056240] ieee80211 phy9: rt2x00_set_rt: Info - RT chipset 3070, rev 0201 detected
[  874.097065] ieee80211 phy9: rt2x00_set_rf: Info - RF chipset 0005 detected
[  874.098679] ieee80211 phy9: Selected rate control algorithm 'minstrel_ht'
[  874.105305] ieee80211 phy9: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
[  874.105432] ieee80211 phy9: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.36
[  874.585117] Under-voltage detected! (0x00050005)
[  875.619091] wlan0: authenticate with c4:ad:34:4f:43:37
[  875.674051] wlan0: send auth to c4:ad:34:4f:43:37 (try 1/3)
[  875.675352] wlan0: authenticated
[  875.676063] wlan0: associate with c4:ad:34:4f:43:37 (try 1/3)
[  875.678676] wlan0: RX AssocResp from c4:ad:34:4f:43:37 (capab=0x431 status=0 aid=1)

```

I've tried to switch the cards from monitor mode to connect them to myLAN:

```
NICS=`ls /sys/class/net/ | nice grep -v eth0 | nice grep -v lo | nice grep -v usb  | nice grep -v int`
 for NIC in $NICS
    do
          echo "Detected NIC devices: $NICS"
                iw dev $NIC info
                iwconfig $NIC | grep $NIC
                ip link set $NIC down
                ip link set $NIC up
                wpa_supplicant -B -D nl80211 -i $NIC -c /etc/wpa_supplicant.conf

```

With (B) no problem, it get up an running:

```
Detected NIC devices: wlan0 
Interface wlan0
	ifindex 9
	wdev 0x600000001
	addr bc:f6:85:66:a1:2b
	type managed
	wiphy 6
	channel 7 (2442 MHz), width: 40 MHz, center1: 2452 MHz
	txpower 30.00 dBm
wlan0     IEEE 802.11  ESSID:"MYLAN"  
Successfully initialized wpa_supplicant
rfkill: Cannot open RFKILL control device
Connected to c4:ad:34:4f:43:37 (on wlan0) SSID: MYLAN 

```

With (A) something seems to go wrong but I cannot find what:
```
Detected NIC devices: wlan0 
Interface wlan0
	ifindex 10
	wdev 0x700000001
	addr 00:c0:ca:98:09:39
	type managed
	wiphy 7
	txpower 30.00 dBm
wlan0     IEEE 802.11  ESSID:off/any  
Successfully initialized wpa_supplicant
rfkill: Cannot open RFKILL control device
Not connected. 

```

htcohio, 2020-02-08:
Hello,

Thank you very much for the info. 

Your problem seems to be related to the issue below.

Long story short, What makes OpenHD (Same EZ-Wifibroadcast 1.6RC6 Core structure) work so well is having end-to-end control over the Wi-Fi packet handling during injection and monitor mode.

Much of Openhd development in the beginning was about opening up full support for the newer RTL8812AU chipsets on 802.11N MCS and MIMO modes. 

We periodically verified atheros AR9271 cards along the way because it's the most common.

Then, at some point during that process of development Ralink cards were discovered to have a few problems for one reason or another.

Looking at your dmesg, initially it actually does look correct because the startup sequence has a couple processes related to binding to the air if users choose to. 

As for A (Atheros) I know we have several driver and firmware patches that can be flaky trying to change them without a hard reset.

Also, mixing different chipsets on the ground for standard use has always been a little problematic. 

Basically, every driver operates a little differently and it's a little difficult to identify based on past experience. 

Perhaps Stephen or Vitaliy would be able to comment with more information. 

https://github.com/OpenHD/Open.HD/issues/220

Summer88, 2020-02-11:
Thank you for explanation, I need to be able to switch from monitor mode to promiscuos.
I've tried to setup the network card at boot, but there aren't any changes, so I'm planning to replace the AR9271.
Thanks, BR

RespawnDespair, 2020-12-04:
This issue appears to be solved, closing.

(side note: WiFi card support has much improved in the latest versions)

closed 2020-12-04 by RespawnDespair