OpenHD/OpenHD #261: How to shut down?

maciek252, 2020-02-09:
It seems that my problems with booting are down to SD cards. Don't know yet if it's the quality of the cards or damaged file systems due to shutting down just by powering off. But... is there any recommended way of shutting down both RPis?

Definitely it shouldn't be done by pulling out the power plug. Maybe there could be a GPIO pin connected to GND during the operation and to 3.3V triggering the system shutdown?

steveatinfincia, 2020-02-19:
There are ways to handle this yes.

The easiest is to grab one of [these specific SD cards](https://www.digikey.com/product-detail/en/atp-electronics-inc/AF4GUD3A-OEM/AF4GUD3A-OEM-ND/5361062), they're industrial cards designed to handle sudden power loss, extreme temperatures, and better endurance to handle flashing new images to the entire card over and over.

When the system stops working after yanking the power, it's generally the card itself (and/or bad power supply) causing the problem rather than the filesystem itself, at least for the root filesystem. Root is an ext4 filesystem with journaling, so as long as the card itself is working properly, if the pi suddenly loses power it'll just recover the journal and fix itself next time the pi boots. The boot partition is another matter, but it's read-only by default at the moment.

QOpenHD has a reboot button and I'm planning to add shutdown as well, which will handle the ground side without requiring additional hardware. Air side requires a little more work but will be possible soon as well.

Lastly there are some nice hardware options that provide safe shutdown via a button, I use the [lifepowered-pi](https://www.crowdsupply.com/silicognition/lifepo4wered-pi-plus) boards that allow PC-style shutdown with an external button, you could do most of that with a simple GPIO as well, and we could add support for that.

estogantsev, 2020-06-29:
Clean shutdown via GPIO is already implemented the RPI firmware. 
Short answer: add "dtoverlay=gpio-shutdown" to /boot/config.txt and connect a button between GPIO 3 (default) and GND
Full description: https://raw.githubusercontent.com/raspberrypi/firmware/master/boot/overlays/README

RespawnDespair, 2020-12-04:
Seems interesting enough to add support in 2.1 I guess.
Clean shutdown will be an issue for all users and this is one viable option.

The other **currently available** option is to use the Power tab in QOpen.HD to shutdown AIR and then GROUND.

See the screen in [this ](https://openhd.gitbook.io/open-hd/advanced-setup/ground-power-monitoring)section in the docs.

I will add this to the 2.1 backlog.

digitalentity, 2021-04-10:
A better alternative IMO is to make sure that boot and root filesystems are always mounted read-only, so yanking the power will not damage anything.

steveatinfincia, 2021-04-10:
The corruption problems with as cards on the pi aren’t in the file system they’re below it, even a read only filesystem won’t help there.

We have boot read only at the moment but only because it’s fat32 and isn’t even journaling. 

The real problem is as cards the pi doesn’t like, usually sandisk because the controllers don’t react well to sudden power loss. Samsung cards seem to do fine in the pi, I cut the power all the time and never see any issues but I used to have those problems all the time with sandisk and no name cards.

It’s something specific to the pi, cards that may work fine in android devices just aren’t always going to work on the pi, either because of voltage issues on the SD lines or the way the SD bus controller interacts with the one in the card.

closed 2022-10-16 by raphaelscholle