OpenHD/OpenHD #271: rtl8811au

kasatka60, 2020-02-16:
rtl8811au chipset (dexp wfa-601). Does not work:
telemetry
cpu boot
cpu temperature
receiver sensitivity.

Russian image Ez-wifibroadcast-1.6RC6+8812au everything works.
[image]

htcohio, 2020-02-18:
Hello,

Yes, this is a fairly well documented and known problem.

The 8811 card is not capable of the same things the RTL8812AU card is. Specifically, packet injection while in monitor mode for the uplink functions that are needed for openHD to function properly. 

We have been able to implement 802.11N "MIMO" STBC/LDPC diversity with the 8812au card end of the performance on 5.8 is extraordinary.  It works perfectly. 

In order to make 8811au work well again, we potentially have to differentiate the two different chipsets and utilise Legacy 802.11bg Rates for the rtl8811au. 

This would still result in some other difficult to track down problems. 

It always ends up requiring many hours of time to get specific wireless cards working properly and, because there's only been two people who have brought this up wanting to use the 8811 cards, the time spend on there would not be beneficial to many users. 

I would highly recommend that you upgrade to rtl8812au as there are small cheap modules available that would provide you all the benefits over be fully functional openHD system.

Here are a couple links to well-known working models. 

1. Here is a $10 low power (approx 100mw max) 8812 module:
https://a.aliexpress.com/_sPDJfJ

2. Here is a $20 High Power 650-1000mw module. (We often refer to this as the "Taobao" card because that's where was first sold). 
https://a.aliexpress.com/_rIjofM

kasatka60, 2020-02-18:
Thanks.
I already ordered rtl8812au 650mw. But in China, the virus has not yet been parceled.

closed 2020-02-18 by kasatka60