OpenHD/OpenHD #278: How to adapt the network card? I have an rtl8822bu

wyml, 2020-03-03:
I have two rtl8822bu network cards. How can I use them?
Or be able to adapt the drive by yourself?
Tank you！

steveatinfincia, 2020-03-13:
As far as I know that one isn't supported at the moment, but if you're interested in making it work you might check to see if the drivers for that chipset will support injection and monitor mode. If so, it *might* be possible to make it work in Open.HD, but a lot of cards that should technically work don't actually work very well in the field.

If you're still interested you might stop by our [Telegram channel](https://t.me/OpenHD_May_2019), someone might already know whether it's worth pursuing :)

wyml, 2020-03-19:
ok thank

closed 2020-03-19 by wyml