OpenHD/OpenHD #282: how to get mavlink from ground pi

rebelboy, 2020-03-18:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Configuration and debug logs**
Please attach the relevant openhd-settings*1.txt and any debugging logs available

**Hardware:**
 - Air side
 -- Pi model
 -- Camera(s)
 -- Wifi card(s)
 -- Other hardware
 - Ground side
 -- Pi model
 -- Wifi card(s)
 -- Ground Control Station(s)
 -- Hotspots - do you connect you GCS using ethernet, wifi and/or usb tethering
 - RC/telemetry uplink
 -- Do you use RC over Open.HD, or a separate rf link?

**Client devices (in case of hotspot/tethering issue)**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - App - FPV_VR, QGC, and version
 - Version [e.g. 22]
 - Connection method

**Additional context**
Add any other context about the problem here.

steveatinfincia, 2020-03-21:
You mean you want mavlink to be sent out of the UART pins on the ground pi?

Check these settings in `openhd-settings-1.txt` on the ground side:

    # >>>>>>>>>>[GROUND SERIAL PORT OUT ENABLE?]--Set to "Y" to enable output of telemetry to serialport on ground Pi
    # (for antenna tracker etc.)
    ENABLE_SERIAL_TELEMETRY_OUTPUT=Y

    # >>>>>>>>>>[GROUND SERIAL PORT/BAUDRATE OUT]-- 
    # Used for ground Pi telemetry output if TELEMETRY_OUTPUT=Y
    TELEMETRY_OUTPUT_SERIALPORT_GROUND=/dev/serial0
    TELEMETRY_OUTPUT_SERIALPORT_GROUND_BAUDRATE=57600

rebelboy, 2020-03-25:
thank you !  sorry for my poor english！

closed 2020-03-31 by rebelboy