OpenHD/OpenHD #286: How to simulate this project on my pc ?

cgrrty, 2020-03-26:
Can I run this project on my pc? if can do,tell me how to do, thanks

steveatinfincia, 2020-04-19:
Hi :)

Are you referring to *simulating* a particular part of the system like working on the OSD/QOpenHD, or are you referring to running the ground side on a PC?

We have been working on adding support for using regular PC hardware as a ground station, will be a few weeks before we merge that, but will definitely be supported soon.

sni-pi, 2020-04-20:
I believe he need a PC as development platform, to avoid the need of having real hardware systems connected with antennas and headache.
If that's the case I'm interested too

steveatinfincia, 2020-04-20:
I do basically all development on this project from a PC, but for actually running the code you still have to do that on the target hardware, there's no way around that. 

For example you can't run `raspivid` on a PC, you have to run it on a Raspberry Pi. You can't run the current ground station video decoder, `hello_video`, on anything else either. 

Other parts of the code are more portable, but still not portable enough to run on a PC. They will always require running them on a Linux system with specific kernel patches applied,and  it's not worth the trouble to mess with doing that on a desktop linux system when you can just run it on a Raspberry Pi.

sni-pi, 2020-04-21:
Something like this seems to be very nice:

https://www.youtube.com/watch?v=oxK4ykVh1EE&feature=youtu.be&utm_source=Robot+Ignite+Academy&utm_campaign=fa05f4776c-EMAIL_CAMPAIGN_LIVE_CLASS_65_COPY_01&utm_medium=email&utm_term=0_5f17ebd98f-fa05f4776c-222366813&ct=t%28EMAIL_CAMPAIGN_LIVE_CLASS_65_COPY_01%29&mc_cid=fa05f4776c&mc_eid=63fdcdf71a

steveatinfincia, 2020-04-21:
If you're mostly looking for a way to get some of the major pieces *for* development set up, I think Luke and Thomas have been working on a virtual machine image with a lot of the core stuff preinstalled. However that won't really "simulate" the project itself, but it would allow you to work on QOpenHD if that's what you were interested in.

cgrrty, 2020-04-25:
Hi 
This is really good news，using regular PC hardware as a ground station.

closed 2020-06-15 by cgrrty