OpenHD/OpenHD #287: Add new feature.

domenic64, 2020-03-29:
New feature?
you can add "air pi"
1) read the form rs232 gps and send it to "gnd pi" to see them on "qonenhd".
2) i2c reads mpu9250 and sends the scroll, the picth, the compass, the data to "gnd pi" to display them on "qonenhd"
Thanks
openhd doesn't have to fly.

steveatinfincia, 2020-04-19:
Hey :)

So you're referring to using OpenHD without a flight controller then right?

What sort of use cases are you thinking of that would not need some sort of controller board handling RC/motors/movement?

underdogest, 2020-04-21:
Cheap or very small FCs often have only one RX pad for RC connection and usualy have no I2C pads at all

domenic64, 2020-04-25:
Hi
I am going to put openhd on hydrofoil.
No RC, No motors.
Video only, minimal telemetry GPS rs232 on airpi0
+ if possible roll yaw pich i2c on airpi0
Thanks

closed 2020-12-04 by domenic64