OpenHD/OpenHD #294: Telemetry data forward on AirPi

bkzspam, 2020-04-16:
Hello, I want to forward telemetry data on AirPi to second serial port (Probably with a USB-TTL) so I can use it with STM32. Is possible to do this?

steveatinfincia, 2020-04-17:
Hi :)

It's possible yes, but not exposed as an option at the moment.

What were you considering using the STM32 for? Perhaps there's a better way to accomplish what you're trying to do :)

bkzspam, 2020-04-17:
> Hi :)
> 
> It's possible yes, but not exposed as an option at the moment.
> 
> What were you considering using the STM32 for? Perhaps there's a better way to accomplish what you're trying to do :)

Short answer: I will use it for sms telemetry over 2G automatic landing gear retraction... (More possibilities)

Explanation:
I will use "crash check" with reply messages that comes(ie. Battery -> Voltage 11.1V 1350Mah remained.). It will make the link more solid, my flight zone almost fully covered by 2G so if UHF link, 2.4Ghz and 5.8Ghz fpv get broken I can recover it. I really have a bad experience about these are, I lost all connections in 8 seconds, thanks to the motor beeps I found it. About 4G, it is has way short coverage and expensive. Data rates aren't really nice too. Plus I don't have phyton experience. So I prefer "C/++".

steveatinfincia, 2020-04-19:
So the telemetry messages you're thinking of are really being generated by the flight controller right? Can you just connect to another UART on the flight controller itself?

bkzspam, 2020-04-19:
> So the telemetry messages you're thinking of are really being generated by the flight controller right? Can you just connect to another UART on the flight controller itself?

I thought it, however it will take an UART. So using everything in RPi's itself can be better. Integration of something like virtual com can be great for these are. Isn't it? Or directly outputting decoded message?

closed 2022-10-16 by raphaelscholle