OpenHD/OpenHD #295: How to use SSH ground pi to control airpi？？？

return97, 2020-04-19:
How to use the ground pi to control the air Pi?

I've tried socket Python script files. But airpi doesn't have any IP address?

I don't know how to use socket to establish the connection between airpi and ground Pi?

Because I want to control the external camera!

Thank you very much!

steveatinfincia, 2020-04-19:
Hi!

The way we will support camera control is through standardized telemetry messages rather than IP sockets, we have a new system for it being worked on at the moment :)

You will be able to control the cameras from the ground station screen (if it's a touchscreen or has a mouse), or from an Android device, or by assigning specific controls to a particular RC channel.

return97, 2020-04-20:
hi 。

How to realize wireless communication between airpi and ground Pi.

I need to control other cameras, standard messages are useless to me.

For example, GPIO of airpi is controlled by ground PI!

shyam98, 2020-04-20:
Hello, 
I'd also like to know if there was a standard method of communicating with the AirPi from the ground pi. My specific use case is I'd like to send information to the AirPi so it can do some pre-programmed autonomous actions / run dronekit scripts while already in the air.

steveatinfincia, 2020-04-20:
> Hello,
> I'd also like to know if there was a standard method of communicating with the AirPi from the ground pi. My specific use case is I'd like to send information to the AirPi so it can do some pre-programmed autonomous actions / run dronekit scripts while already in the air.

You wouldn't run anything like that on the air side, you would use normal telemetry messages via the ground station, generally through the hotspot.

steveatinfincia, 2020-04-20:
> hi 。
> 
> How to realize wireless communication between airpi and ground Pi.
> 
> I need to control other cameras, standard messages are useless to me.

What do you mean by other cameras? Open.HD isn't a general platform for controlling 3rd party equipment, but if you describe what you're actually doing instead of asking about general communications perhaps we can help.

> For example, GPIO of airpi is controlled by ground PI!

We have a camera switching system for the ArduCam camera switcher, and we have generic GPIO support you can control through custom Mavlink messages.

You can also run QOpenHD on Android, enable the GPIO widget in the settings area, and then directly control some of the air side GPIOs from the screen.

shyam98, 2020-04-24:
> You wouldn't run anything like that on the air side, you would use normal telemetry messages via the ground station, generally through the hotspot.

Gotcha, ideally, I'd like to send telemetry messages from the ground station when the drone is out of range of a hotspot, but still in the range of telemetry. Any ideas on how to do this? I was also considering using something like LoRa directly to the AirPi from the ground Pi to send this information. Perhaps that is something  @return97  might also be interested in looking into for their application. 
Thanks!

steveatinfincia, 2020-04-25:
> Gotcha, ideally, I'd like to send telemetry messages from the ground station when the drone is out of range of a hotspot, but still in the range of telemetry.

By hotspot do you mean the Open.HD link between ground<->air?

The general idea of Open.HD is to have full communication with the drone all the time, even out to extreme range (we have test flights to 70KM).

If you're using Open.HD for telemetry (the default setup), and you choose your wifi card and antenna carefully (and possibly use an antenna tracker on the ground), you might actually run out of battery power long before you go out of range, even for the video stream.

shyam98, 2020-04-25:
My apologies, I believe I'm being confusing haha. 

Basically, I have dronekit scripts that I want to invoke while the drone is at some distance in the air (~500m). I was wondering if there was a way to do this using OpenHD. I'd like to run the dronekit scripts off of the airPi which would also mean I'd have to invoke commands on the AirPi from the ground station using OpenHD.

steveatinfincia, 2020-04-27:
500m should be easily within range, have you tested the distance yet?

If you're still within range, you can just run the dronekit scripts on the ground pi and talk to the drone directly that way. It doesn't have to run on the air pi for that (at the moment there's no easy way to do this on the air side anyway).

To make it work right now you can use the mavlink-router port that QOpenHD normally uses (if you aren't using QOpenHD on the ground pi). All you would need to do is tell your dronekit script on the ground pi to connect to UDP 127.0.0.1:14550, that will give you 2-way communication with the drone.

shyam98, 2020-04-27:
That sounds like exactly what I need, thank you for all the help @steveatinfincia !!! I tried Ez-wifibroadcast long ago on a custom build but now I'd like to try to do something a bit more extensive. Did not have Open.HD back then, you all are doing great work!

return97, 2020-04-28:
The cloud PI sends an integer or floating-point data to the air pi

Then the air PI receives the data and gives the corresponding instructions! That's what I think

I tried socket UDP but the air PI never received any data!

I don't know how to use socket UDP to connect the ground PI and air PI!

RespawnDespair, 2020-12-04:
It appears the original question has been answered.

The last communication was some time ago.
Closing this issue for now.

If you want to continue the conversation, please reopen the issue.

closed 2020-12-04 by RespawnDespair