OpenHD/OpenHD #299: About the RTSP address of other cameras?

return97, 2020-04-24:
I want to use raspberry pi to connect other RTSP video streams to the camera!

I don't know it's my parameter problem? RTSP address is not supported!

He seems to be using the TCP protocol to connect!

rtsp://192.168.1.240:8557/PSIA/Streaming/channels/2?videoCodecType=H.264

This is the address!

Please help me, thank you!

Also about the parameter settings!

steveatinfincia, 2020-08-25:
Hey :)

The address of the ethernet connection on the pi is in the 192.168.0.x, so a camera in the 192.168.1.x range will not work at the moment. Is there are way you can change that address on the camera?

RespawnDespair, 2020-12-04:
The last communication with the original poster was some time ago.
Closing this issue for now.

If you want to continue the conversation, please reopen the issue.

closed 2020-12-04 by RespawnDespair