OpenHD/OpenHD #301: The air pi will stop working when you turn it on for a few minutes (Raspberry Pi4b)

Windtweet, 2020-04-26:
**Describe the bug**
The air pi (Pi4b) run the software "Open.HD-nightly-2020-4-13-buster" and it can work properly in few minutes. Although I don't touch it, it still will stop work after a few minutes.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on the Ground pi and the Air pi (Both of them are Pi4b)
2. The screen will show the video
3. Wait for a few minute
4. The Ground pi still running but no signal

**Expected behavior**
It will work properly
**Configuration and debug logs**
I haven't change any texts.

**Hardware:**
 - Air side
 -- P4b
 -- 500W pixel camera   V1
 -- Wifi card(s)  ASUS  usb  AC56
 -- 
 - Ground side
 -- P4b
 -- Wifi card(s)  ASUS  usb  AC56
 -- None
 
**Client devices (in case of hotspot/tethering issue)**
 - None
**Additional context**
None

steveatinfincia, 2020-04-26:
Hi :)

In this case it's the ground side that has an issue, there's a bug being investigated right now and a fix should be available in the next few days.

Sorry for the trouble!

Windtweet, 2020-04-27:
> Hi :)
> 
> In this case it's the ground side that has an issue, there's a bug being investigated right now and a fix should be available in the next few days.
> 
> Sorry for the trouble!

Hallo,
I tried to use a different usb wifi card on the air pi (A usb 2.4G wificard not the ASUS AC56) and don't change any thing on the ground pi (USB ASUS AC56). Then it works. I tested it for nearly an hour and there's no problem on it. So I guss the problem is on the ASUS AC56 card. Maybe is the driver? I'm not sure. 
I'm from china so I might can't reply your answer in time and I'm glad to hear form you.

steveatinfincia, 2020-04-27:
Ok, you might have a different problem than I thought.

How is the AC56 wired? May need to wire power to it separately, bypass the USB port. This is a common problem with rtl8812au cards like your AC56.

blacksky008, 2020-05-09:
我之前也遇到过这个问题，就是运行一段时间后出现死机状态，我的解决方法是用5v3A的稳压电源给地面的pi供电就好，到目前为止没有再发生画面静止（卡死）的情况。希望对你有帮助

Windtweet, 2020-05-09:
> 我之前也遇到过这个问题，就是运行一段时间后出现死机状态，我的解决方法是用5v3A的稳压电源给地面的pi供电就好，到目前为止没有再发生画面静止（卡死）的情况。希望对你有帮助

谢谢啦，我这里目前在天空端还有地面端USB口加强供电后测试了两个小时都没有问题，问题解决了。麻烦再问一下，地面端安装两个接收器有多大必要，效果有多大提升？谢谢！

Windtweet, 2020-05-09:
> Ok, you might have a different problem than I thought.
> 
> How is the AC56 wired? May need to wire power to it separately, bypass the USB port. This is a common problem with rtl8812au cards like your AC56.

Thank you very much! The problem has been solved.

blacksky008, 2020-05-09:
> > 我之前也遇到过这个问题，就是运行一段时间后出现死机状态，我的解决方法是用5v3A的稳压电源给地面的pi供电就好，到目前为止没有再发生画面静止（卡死）的情况。希望对你有帮助I have encountered this problem before, that is to run for a period of time after the failure of the state, my solution is to use 5v3A power supply to the ground Pi power, so far there is no screen static (stuck) situation. I hope it helps
> 
> 谢谢啦，我这里目前在天空端还有地面端USB口加强供电后测试了两个小时都没有问题，问题解决了。麻烦再问一下，地面端安装两个接收器有多大必要，效果有多大提升？谢谢！
> 
> Thank you, I am currently in the sky and the ground side of the USB port to strengthen the power supply after two hours of testing no problem, problem solved. Please ask again, how necessary is the installation of two receivers on the ground end, how much improvement? Thank you!

我目前测试的也是天空端和接收端分别只用了一个AC56网卡，没有上天测试，据说用2个网卡可以提升带宽，接收的信号会比较好，具体的也没有进行测试。现阶段我还处于在地面测试，飞机也在赶工中QAQ。我现在还没能显示出OSD，在查阅相关资料解决问题中。如果你有好的方法，希望也可以帮助我~

Windtweet, 2020-05-09:
> 
> 我目前测试的也是天空端和接收端分别只用了一个AC56网卡，没有上天测试，据说用2个网卡可以提升带宽，接收的信号会比较好，具体的也没有进行测试。现阶段我还处于在地面测试，飞机也在赶工中QAQ。我现在还没能显示出OSD，在查阅相关资料解决问题中。如果你有好的方法，希望也可以帮助我~
我是这样的，上个星期我出门试了一下距离，AC56一个平板一个全向，走了300m就花屏了，有点不像是它应有的水平，正在调教中，听他们说两个接收会好很多，就是100+RMB的价格有点……所以想先问一下你，谢谢啦！
你比我的进度快多了，我恐怕得今年暑假才能有时间去弄飞机上的事情，飞控还没买哈哈哈。如果osd没出来的话建议再看一下作者的文章https://github.com/OpenHD/Open.HD/wiki/Software-~-Advanced-~-Telemetry-&-OSD应该有帮助，我听他们说连接飞控的串口线时不要接5伏正极，只接TX和RX还有GND三根，否则会烧树莓派。还有TX和RX线要反过来接。我目前可能也就可以帮到这里了，如果还有问题欢迎讨论。
http://bbs.5imx.com/forum.php?mod=viewthread&tid=1554253&extra=page%3D1&page=1
这个网页可以去看看，我就是被他安利过来的。
加油！

Windtweet, 2020-05-09:
> 我目前测试的也是天空端和接收端分别只用了一个AC56网卡，没有上天测试，据说用2个网卡可以提升带宽，接收的信号会比较好，具体的也没有进行测试。现阶段我还处于在地面测试，飞机也在赶工中QAQ。我现在还没能显示出OSD，在查阅相关资料解决问题中。如果你有好的方法，希望也可以帮助我~

我是这样的，上个星期我出门试了一下距离，AC56一个平板一个全向，走了300m就花屏了，有点不像是它应有的水平，正在调教中，听他们说两个接收会好很多，就是100+RMB的价格有点……所以想先问一下你，谢谢啦！
你比我的进度快多了，我恐怕得今年暑假才能有时间去弄飞机上的事情，飞控还没买哈哈哈。如果osd没出来的话建议再看一下作者的文章https://github.com/OpenHD/Open.HD/wiki/Software-~-Advanced-~-Telemetry-&-OSD应该有帮助，我听他们说连接飞控的串口线时不要接5伏正极，只接TX和RX还有GND三根，否则会烧树莓派。还有TX和RX线要反过来接。我目前可能也就可以帮到这里了，如果还有问题欢迎讨论。
http://bbs.5imx.com/forum.php?mod=viewthread&tid=1554253&extra=page%3D1&page=1
这个网页可以去看看，我就是被他安利过来的。
加油！

TheNewJavaman, 2020-08-25:
> > Ok, you might have a different problem than I thought.
> > How is the AC56 wired? May need to wire power to it separately, bypass the USB port. This is a common problem with rtl8812au cards like your AC56.
> 
> Thank you very much! The problem has been solved.

If there are no further problems, this issue should be closed.

closed 2020-08-25 by steveatinfincia