OpenHD/OpenHD #302: RC Simulator

Summer88, 2020-04-28:
Sometime ago, I've opened this issue for telemetry simulation, by this you test if telemetry came from airpi to grounpi OSD:
OpenHD/Open.HD#198

Now I wondering is there a similar way to check RC functionalities without real hardware connected?

Thanks, BR

TheNewJavaman, 2020-08-25:
Currently there's no official method to do this. Would you like to keep this issue open as a feature request instead?

RespawnDespair, 2020-12-04:
It appears the original poster has not commented in  a while.

Please reopen the issue if you want to continue the conversation.
I must admit I'm not quite sure what scenario the original question is targeting. The very basis of the RC control is having a joystick connected to the ground so I assume it is to test whether the RC packages arrive at the AIR unit? One would need to monitor the MAVLink stream coming out of the serial port in the AIR. 

Again closing due to inactivity for now. But feel free to reopen.

closed 2020-12-04 by RespawnDespair

TheNewJavaman, 2020-12-04:
To clarify @RespawnDespair : A way to ensure that a controller is properly connected to the ground station without having to fly actual hardware. I would also like this feature; can you reopen this issue as a feature request?