OpenHD/OpenHD #303: Will this program be supported on openwrt in the future

lsy512, 2020-04-28:
Will this program be supported on openwrt in the future？

webbbn, 2020-04-28:
Depends on what you mean by "this program" and "supported".

A version of wifibroadcast already runs on a version of openwrt: https://github.com/libc0607/YJSNPI-Broadcast

IMO it's mostly a matter of finding the right hardware to run it on. The wifi adapter has to support wifibroadcast with sufficient Tx power. It also needs (IMO) to be in a small enough package to make it compelling compared to other hardware.

I have a couple of nice boards sitting next to me that run openwrt and I'm pretty sure they would support wifibroadcast, but the source to build the firmware is not readily available, so building a modified openwrt that will run on them would be a challenge.

Is there some hardware platform that you have in mind?

lsy512, 2020-04-29:
That's exactly what I want. Thank you very much.I want to try to run on the platform of qca9531.

RespawnDespair, 2020-12-04:
It seems the original question has been answered, I will close this issue.

If there are any follow up questions, don't hesitate to create a new issue!

closed 2020-12-04 by RespawnDespair