OpenHD/OpenHD #304: Controlling Gimbal through OpenHD

shyam98, 2020-04-30:
Hello,

I was wondering if there was way to control servos and gimbals through pwm gpio pins of the AirPi. I'd like to be able to control a gimbal that is not connected to the flight controller. I see that DroneBridge has some plugin support and there is a [plugin ](https://github.com/DroneBridge/DroneBridge/tree/master/plugins/rc_to_gpio_mapper)for this type of feature so I was wondering if there was something similar for Open.HD 

Thanks!

steveatinfincia, 2020-04-30:
We do have simple remote GPIO support, pwm is certainly possible I’ll do some testing :)

shyam98, 2020-05-01:
Thanks @steveatinfincia ! This would add some more testing abilities to my setup before I get it in the air.

RespawnDespair, 2020-12-04:
We will add this issue to the backlog for 2.1
The architecture in 2.1 allows for easier addition of a module for this.

It has been (and still is) very busy getting the 2.0 to a stable state, so it might be some time before any effort can be spent on this