OpenHD/OpenHD #311: Open.HD on Nvidia TX2

neom89, 2020-05-15:
Is there a way to use Nvidia TX2 or any similar grade SBC for video transmission instead of RPi?

steveatinfincia, 2020-05-15:
Yes, we’re adding support for the jetsons in the coming weeks :)

neom89, 2020-05-16:
That's great. Thank you!

steveatinfincia, 2020-05-19:
Which carrier are you using with your TX2?

neom89, 2020-05-30:
I have developed my own carrier board for tx2.

neom89, 2020-05-30:
Will 2.0 release have support for tx2?

steveatinfincia, 2020-06-08:
> Will 2.0 release have support for tx2?

2.1 will

kidspro88, 2020-06-11:
How about jetson nano? @steveatinfincia

steveatinfincia, 2020-06-11:
Yep, nano too. A few users are testing a tiny carrier board for the Nano designed for drones.

[image]

kidspro88, 2020-07-28:
When will you release 2.1 @steveatinfincia Thanks?

flythecode, 2020-08-07:
> Yep, nano too. A few users are testing a tiny carrier board for the Nano designed for drones.
> 
> [image]

what carrier boards are you testing, the form factor looks nice on the one in the pic.

steveatinfincia, 2020-08-13:
> When will you release 2.1 @steveatinfincia Thanks?

Betas are expected in about 2 weeks from now.

ghulands, 2020-08-18:
where can we get one of those carrier boards?

ghulands, 2020-08-26:
@steveatinfincia let me know if there's any tasks you need a hand with and I can help contribute.

RespawnDespair, 2020-12-04:
Just an update, 2.1 is not out for public development yet, as Stephen mentioned, the focus is on 2.0 stability now.
Any help in the development will be greatly appreciated when we buckle down on 2.1 development.
I will add this issue to the backlog for 2.1 so we will pick it up and we will notify people if any help is required.

steveatinfincia, 2020-12-07:
We're not far off though, I'm doing most development for 2.1 on a Jetson nano, all of the initial setup and hardware detection works great, just working on getting it all merged as quick as I can so I can build test images for people.

kidspro88, 2021-01-13:
Will you provide a docker image or an image on Jetson Nano?

millerda, 2021-03-10:
Any update on Jetson Nano support yet?

hpuxionghao, 2021-08-31:
> We're not far off though, I'm doing most development for 2.1 on a Jetson nano, all of the initial setup and hardware detection works great, just working on getting it all merged as quick as I can so I can build test images for people.

hi @steveatinfincia  when 2.1 can be released ,I hope Open HD can be used on Jetson nano , thanks a lot

closed 2022-10-16 by raphaelscholle