OpenHD/OpenHD #312: [Feature request] A Raspberry CM3 carrier board

digitalentity, 2020-05-16:
It would be nice to have a RPI CM3 carrier board with the following features:

1. Several sets of mounting holes - to allow effortless assembly on more-less standard frames with 30.5mm mounting holes.
2. Built-in BEC with 6-8S input voltage capability (plus a TVS to protect the board) - to provide a stable and reliable power to RPI, camera and wifi card
3. USB pads to solder a cable connecting a wifi dongle
4. Camera connector in a suitable place (narrow edge of the board)
5. RX/TX pads for both UARTs for easy telemetry connection to the flight controller
6. SD-card slot (with eMMC for storing the firmware, external SD-card can be used for in-air recording (simply dumping raw stream from x264 encoder before fec/interleave would be sufficient IMO, it can be converted to commonly used container later, after ejecting the card)

steveatinfincia, 2020-05-19:
One of our users (Nikita Gusakov) has been working on a custom CM3 carrier board, I'll let him know you are interested and perhaps this issue can be used for more structured discussion than we have been having in telegram.

> SD-card slot (with eMMC for storing the firmware, external SD-card can be used for in-air recording (simply dumping raw stream from x264 encoder before fec/interleave would be sufficient IMO, it can be converted to commonly used container later, after ejecting the card)

This is going to be possible with the new camera service in OpenHD 2.1 or 2.2 :)

RespawnDespair, 2020-12-04:
Since the release of the new CM4 we have formed a group that is actively developing a Carrier for that board.
I know David Sas (on Telegram) has a working CM3 carrier with some lovely features. Perhaps you can contact him to see if it is possible to obtain one of those.

Since there is already a big effort going on in for the CM4 and all of your features are actually taken into account in that project i will close this issue. We will report any progress on the carrier in the main telegram group.

closed 2020-12-04 by RespawnDespair