OpenHD/OpenHD #316: Streaming from GoPro

maciek252, 2020-05-19:
Hi,
I want to use a GoPro located on a gimbal for providing video for Open.HD. This is however not easy as the HDMI cable is stiff and the gimbal must deal with this extra load. Moreover, an adapter from HDMI to RPi is needed. Would it be possible to receive a video stream from GoPro over WiFi? Even with worse image quality, it would be compensated by very good stabilization (I'm not sure about HyperSmooth - is this present in on-line HDMI? For sure gimbal would work). Probably an extra WiFi adapter will be required? Can Open.HD handle this? The GoPro API is more or less known:

https://github.com/KonradIT/goprowifihack
https://github.com/KonradIT/GoProStream

steveatinfincia, 2020-05-20:
There have been a couple people asking about this in the past but it's generally not a good idea to have other transmitters on the air side, particularly in the same bands as the main wifi card. Even if you could avoid it causing interference, the latency would probably be outside the range you would want for video.

You can definitely get really lightweight HDMI cables though, there are some that use the same flat ribbon as the pi cameras, a lot of people do that with regular FPV/action cameras.

It'll be easier to experiment with stuff like this when the new camera service is merged, doing things like snapping still shots is an explicit feature we're going to support, as is the ability to use multiple sources of different types so it'll be easier to plug-in another type. 

It would be pretty complicated right now even just on the wifi side, as the code that handles the internal wifi card on the pi doesn't run at all on the air pi, and is designed for setting up a hotspot rather than joining one created by another device (assuming that's what the gopro does, if not you probably could get away with enabling the hotspot on the air side and making the gopro join it, same interference issues mentioned above apply though).

Right now the video pipelines are a mix of raspivid and gstreamer, and use 2 different transmission methods depending on several settings, it's a little complex to deal with but it's going away soon.

If you really wanted to try it soon though I would say first check and see how well the stream itself works with a gstreamer pipeline.

maciek252, 2020-05-20:
Many thanks for the answer :)

One could run the gopro-RPi link on 5.8GHz, the air-ground link on 2.4 GHz and the drone RC link on 868 kHz, this in principle should work fine. Or different 2.4 GHz frequencies for both RPi comms.

I'll try to trigger the GoPro stream and check how it works, and how to enable it. In particular, if it involves Bluetooth. On GoPro 8, I've already noticed that the WiFi connection needs to be triggered by BT, contrary to GoPro 7.

maciek252, 2020-05-21:
I've managed to run the high-quality streaming from GoPro 8 Black. First I had to connect to the cam using the phone app. I had to borrow the phone from my wife as mine has too old Android version for running the GoPro app :) I've also configured a nginx+rtmp server running linux ubuntu. I've used the app to type the url, the wifi network, and then the streaming started. I've connected with vlc. There is a really big (but constant) lag, about 3 seconds, but the quality of the image is very good. After establishing the connection the phone can be switched off and it still runs, so it's no used for retransmission. Maybe one could initialize the whole connection without the phone app, but even the described scenario is perfectly feasible and I hope worth implementing to Open.HD

pilotnbr1, 2020-05-21:
3 seconds is really really high. I would guess most flyers arent interested in anything over 300ms of latency.... Also one more wifi signal on a machine already emitting tons of rf is not ideal, even if they are on separate bands.

I really think you would be better off exploring more flexible cable connections that your gimbal can tolerate

maciek252, 2020-05-21:
For me 3 seconds is perfectly fine assuming that the telemetry data is updated on-line (even if not in sync with the video). I usually fly pre-programmed routes and expect the video only to confirm that everything is fine. 

I've ordered already 4 HDMI cables and all of them have proved to be not flexible enough. The gimbal really doesn't like them... Also they are not very resistant to crashes or even hard landings. And on a plane with a gimbal even a relatively good landing puts some stress on the camera. It's easy to damage the HDMI socket.

ijustwant, 2020-05-23:
Just disable the video stabilize int the gopro and the lag is gone.

RespawnDespair, 2020-12-04:
This seems like a really interesting feature.
I would guess that the RTMP stream could be fed to the Air Raspberry Pi using the same custom command lines as the USB and IP cameras do now.
The setup and initialization do sound a bit fiddly.

I think it would be a nice to have feature in 2.1, so I'll add it to that project for now.
We will probably come back with more questions when we start work on this...

closed 2023-01-31 by raphaelscholle