OpenHD/OpenHD #318: How to build Open.HD?

maciek252, 2020-05-20:
Is there any description available how to build the project and do some local modifications? Say I would want to:
-do a "hello world" stuff such as changing a letter in a label showing on the screen
-choose a different source of video to be transmitted. I'm planning some GoPro experiments, but to begin with I would be happy to use a static photo image, just to learn how the whole thing works.

There are some remarks for developers on the wiki, but I can't see a toolchain and build process description.

steveatinfincia, 2020-05-20:
Hey :)

There's build instructions in the [builder repository](https://github.com/OpenHD/Open.HD_Image_Builder). It's a long process at the moment, we're soon moving to packages instead of a complex builder so things will be much easier to deal with then.

> -do a "hello world" stuff such as changing a letter in a label showing on the screen

The old OSD is in the wifibroadcast-osd subdirectory of this repo, the new one is in the [QOpenHD repo](https://github.com/OpenHD/QOpenHD). The new stuff is far easier to work on and the old OSD only works on the older pi models.

> -choose a different source of video to be transmitted. I'm planning some GoPro experiments, but to begin with I would be happy to use a static photo image, just to learn how the whole thing works.

The way to do this at the moment would be to use gstreamer and figure out which pipeline elements you need to receive the stream from the camera. We're about to replace the mix of gstreamer and scripting with a high level camera service with multiple "plugins" for different sources, sounds like what you're doing would be easier at that point but you can still do some testing with gstreamer if you like.

maciek252, 2020-05-20:
Thanks! Indeed it seems a complex challenge. I'm thinking how to get extra 30 GB :). If I want to change something in QOpenHD, would it work to call the script, which will clone it, and then change the code in the respective folder, and compile it with make? Or will the script initiate compilation on its own?

Which part is responsible for AirPi selection of image? Assuming that I would want to transmit photos, possibly taken by a GoPro (as described in another thread?)

steveatinfincia, 2020-06-08:
Sorry I missed this for so long!

> If I want to change something in QOpenHD, would it work to call the script, which will clone it, and then change the code in the respective folder, and compile it with make? Or will the script initiate compilation on its own?

You need the Qt toolchain installed, and that is slightly different depending on which platform you're running it on. There are build instructions in the QOpenHD repository.

I added an example widget the other day for someone so you can use that as a reference for how to create them.

> Which part is responsible for AirPi selection of image? Assuming that I would want to transmit photos, possibly taken by a GoPro (as described in another thread?)

That is a combination of things (and they're not all in one place at the moment). The earliest selection happens in the `.profile` file which ends up in `/root/` on images. That detects whether a pi camera, HDMI-CSI board, or Veye camera is connected. It also looks for a file called `air.txt` in the boot partition, which forces the system to boot in air mode. That is necessary at the moment because IP and USB cameras are not autodetected, but they will be in 2.1.

TheNewJavaman, 2020-08-25:
Are you still having issues?

steveatinfincia, 2020-08-25:
This should be significantly easier now, all I do at this point is clone the builder repo and run `./build.sh pi buster`, everything is packaged so on a fast connection it only takes 10-15 minutes at the most, significantly faster if you have an SSD on the build machine.

If you're working on something you are *far* better off just cloning the source for the component you're using directly on a raspberry pi and running the package script to build it (a few of them require installing ruby-dev, then running gem install fpm). The image building stuff isn't a way to change things on the system anymore but just to preinstall the packages.

closed 2020-08-25 by steveatinfincia