OpenHD/OpenHD #319: yocto?

zygmunt, 2020-05-20:
Hi,
Did anyone assume (or try) use yocto as a base system for building OpenHD image?

steveatinfincia, 2020-05-20:
Hi! :) 

I've done some stuff with Yocto before yes, but we're in the process of moving toward packaging and preinstalling things on stock images rather than using a complex builder anyway. The "builder" will end up just being a list of packages to install.

zygmunt, 2020-05-20:
Yes, yocto is complex (or maybe not:)).
But the idea is to reuse what is implemented, and write receipes for something what have to be:)
I can't promise, but I have some plans to dive again into OpenHD, maybe I will be able to build something with yocto.

steveatinfincia, 2020-05-20:
> Yes, yocto is complex (or maybe not:)).
> But the idea is to reuse what is implemented, and write receipes for something what have to be:)
> I can't promise, but I have some plans to dive again into OpenHD, maybe I will be able to build something with yocto.

There won't actually be anything to build when creating images is what I mean, we're going to be packaging things independently (either tarballs or debs, probably debs) so they can be preinstalled on different kinds of images and then upgraded in-place. For that we really just need a simple shell script that runs apt and that's it.

zygmunt, 2020-05-20:
> we're going to be packaging things independently (either tarballs or debs, probably debs)
Nice(Y)

steveatinfincia, 2020-08-25:
Closing as things are packaged now, and we're essentially using stock images for the most part on each platform rather than doing much to customize the image itself.

closed 2020-08-25 by steveatinfincia